Traumflug/Teacup_Firmware #200: capability to control the ready to use cutting modules such as auto focus laser head or plasma THC

mkeyno, 2016-03-22:
hi , I was wondering if this firmware have capability to control the ready to use cutting modules such as auto focus laser head or plasma THC , 
I saw couple of effort on Marlin Due enthusiast to check output sensor the cutting tool and control the stepper motor the module to control height during the printing 
any comments really appreciated

Traumflug, 2016-03-22:
Any voltage consuming devices (fan, lights, lasers, ...) can be added like a heater, with or without temperature sensor.

Tracking a sensor with a stepper motor would be a new feature, code is welcome.

mkeyno, 2016-03-22:
.Actually I made 120 watt laser CNC with RAMPS and modified Marlin by[ TurnkeyTyrann](https://github.com/TurnkeyTyranny)y which is equipped with BT wireless link , I used to repetier host for cutting and engraving. Code has minor defect and unfortunately not updated any longer , however I'm moving to next step and find many industrial laser CNC have module which is keep the cutting tool; in readiness state , for example THC module check the correct distance from material constantly ,
also I find many modified codes here has own approach and not follow the same standard way as 3d printer developers goes, hence I start sending the note to Marlin and repetier developer to find any enthusiast people to such project and make standard modified version for cutting CNC machine
I really appreciated to have your comments

Traumflug, 2016-03-22:
Doesn't exactly sound like you're about to fire up your code editor. :-)

closed 2016-03-26 by Traumflug