Traumflug/Teacup_Firmware #201: Pullups for Endstops

Wurstnase, 2016-03-26:
https://github.com/Traumflug/Teacup_Firmware/tree/pullups_for_endstops

Just has the issue that I can't set up my endstop properly. The enstops are NC and connected to GND.

Traumflug, 2016-03-26:
Hmm. There is such functionality already. It's *USE_INTERNAL_PULLUPS*.

Maybe it's not too apparent, because it's a bit hidden in ```endstops_on()``` and ```endstops_off()```. I added the ability to turn endstops on on demand only, because when using a milling spindle as one endstop contact and the metal workpiece as other endstop contact, one has a simple touch probe. And in such a setup, the pulling voltage causes a constant current from milling bit to workpiece, causing the milling bit to slowly erode. There are no flying sparks, of course, but PCB milling tools wear out double as fast as usual.

Does this 'endstops on on demand only' work for you? If not, what's missing?

That said, Gen7-ARM doesn't need pullups, but pulldowns, so I have a local patch here introducing PULLUP_ON(), PULLDOWN_ON() and PULL_OFF().

Wurstnase, 2016-03-26:
> Hmm. There is such functionality already. It's USE_INTERNAL_PULLUPS.

This will not work. When endstops_off() they are triggered in that case.