Traumflug/Teacup_Firmware #202: Z late enable

Wurstnase, 2016-03-26:
In normal mode the z-axis is enabled and disabled after each move.
My z-axis spindle has no self locking and will lose its position.

Traumflug, 2016-03-26:
I'm pretty sure we had this issue a number of times already, so sorry it isn't fixed already :-) And yes, code introducing a config flag would be welcome. Ideally together with a corresponding patch for Configtool.

Wurstnase, 2016-03-26:
> Ideally together with a corresponding patch for Configtool.

done

Traumflug, 2016-03-27:
Excellent! Thank you for the contribution. Picked this to *experimental* and removed the topic branch.

I hope you don't mind me renaming Z_LATE_ENABLE to Z_AUTODISABLE. I hope users understand this better.

Wurstnase, 2016-03-27:
You're welcome. 
Yes, Z_AUTODISABLE is maybe a better naming.

closed 2016-03-27 by Traumflug