Traumflug/Teacup_Firmware #207: Developing Configtool

Wurstnase, 2016-04-11:
Hi all,

I'm stuck a bit with the configtool. I try to implement some radiobuttons. 2 comes from a list and one is a dummy like 'None' in Acceleration Types. I want to setup the dummy as default, but I don't get it to work.
```
    self.tempCtrlKeys = ['BANG_BANG', 'PID_AUTOTUNE']

    vsz = wx.BoxSizer(wx.VERTICAL)

    b = wx.StaticBox(self, wx.ID_ANY, "Temperature Control")
    b.SetFont(font)
    sbox = wx.StaticBoxSizer(b, wx.VERTICAL)
    sbox.AddSpacer((5, 5))
    style = wx.RB_GROUP
    self.rbPID = wx.RadioButton(self, wx.ID_ANY, "PID", style = style)
    self.rbPID.SetFont(font)
    self.rbPID.SetValue(True)
    sbox.Add(self.rbPID, 1, wx.LEFT + wx.RIGHT, 16)
    sbox.AddSpacer((5, 5))

    style = 0
    for k in self.tempCtrlKeys:
      rb = self.addRadioButton(k, style, self.onTempCtrlSelect, b)

      sbox.Add(rb, 1, wx.LEFT + wx.RIGHT, 16)
      sbox.AddSpacer((5, 5))

    vsz.Add(sbox, 1, wx.LEFT, 40)

    cb = self.addCheckBox('E_ABSOLUTE', self.onCheckBox)
```

jbernardis, 2016-04-12:
The way radio buttons work is that the groups are defined through the way you set the style.  If the style is set to wx.RB_GROUP, a new radio button group is created.  You then set the style to 0 as you add radio buttons to this group - which is exactly what you are using.

Within a group, only one can be set to True.  I can see in the above code that you are setting the value of rbPID to True, which should "press" that button.  I can't see what happens beyond your code excerpt  though. If you set ANY of the buttons in this group to True, it will cause rbPID to be reset back to False.  In other words, the initial state of a group of radio buttons is determined by the LAST button you set to True, not the first.

If you are not subsequently setting any of these buttons to true, then I don't see what could be causing the problem.  If you could provide some additional code context, or a better explanation of what is actually happening, I could probably dig a bit deeper.

Wurstnase, 2016-04-12:
Thanks for your comment.

I'm not very familiar with this. I'm using nearly the same code from the [accelerationpage](https://github.com/Traumflug/Teacup_Firmware/blob/master/configtool/accelerationpage.py#L27-L45).
I put the SetValue before the loop, behind the loop. Anytime the second element is 'true'.

Wurstnase, 2016-04-12:
I think I find the issue.

```
  def insertValues(self, cfgValues):
    Page.insertValues(self, cfgValues)

    self.checkBoxes['LOOKAHEAD'].Enable(False)
    self.textControls['ACCELERATION'].Enable(False)
    for tag in self.accTypeKeys:
      if tag in cfgValues.keys() and cfgValues[tag]:
        self.radioButtons[tag].SetValue(True)
        if tag == 'ACCELERATION_RAMPING':
          self.checkBoxes['LOOKAHEAD'].Enable(True)
          self.textControls['ACCELERATION'].Enable(True)
```

It's in the very end of the xxxpage.py.

closed 2016-04-12 by Wurstnase