Traumflug/Teacup_Firmware #213: We've been awarded with a Bounty!

Traumflug, 2016-04-30:
No no, this isn't spam from the internet underground, it's real! @bobc is exited enough about Configtool to have it choosen to fulfill his $50 bounty requirements. I'm pleased.

See https://github.com/bobc/RepRap-meta-proto/issues/1#issuecomment-215954391

@jbernardis, you're the one who did most of the work and who made Configtool possible at all. I think it's you who deserves the price by closing the issue over there.

What do others think?

jbernardis, 2016-04-30:
> See bobc/RepRap-meta-proto#1 (comment)

Wow!!

I'm honored.  I will defer to the wishes of the group before I claim the
bounty.   Although I agree that I did the majority of the work, a lot of
people have touched the code - some significantly - since I delivered it.

Jeff

phord, 2016-05-01:
No arguments here. I think @jbernadis drove this feature eagerly and flexibly from start to finish.