Traumflug/Teacup_Firmware #214: HD44780, 1602A and similar displays

Traumflug, 2016-05-02:
A peek onto my desk:

[image]

Display is a "1602A V2.0", which happens to be the same as a HD44780, as far as I could find out. It was extracted from an Arduino shield from DFRobot, which in turn was a donation by @jgrjgr. Thanks, Jolly.

jgrjgr, 2016-05-04:
ah sweet , that should cover just about most of the lcd controllers on the market , do you have any ideas on creating a modular lcd menu for different setups? eg 3d printing, cnc milling , pick n place

Traumflug, 2016-05-04:
So far I have no idea on what content to put there. In a few days I'll be done, then you can do what you like with code as simple as
```c
display_writestr_P(PSTR("static text here"));
// and / or
sendf_P(display_writestr, PSTR("format here"), variable, variable, ...);
```

jgrjgr, 2016-05-30:
the menu system i believe for teacup should be a painfully simple as it needs to be, personally while i really like the way they have done the menu system in code for marlin, it's too complex for what you really need to do,   ideally all we really need in a menu system is temperatures of the hotend and bed displayed and enough of a menu system do that a file from an sdcard can be selected anything after that is a luxury especially if the one of the goals with teacup is small program size

Traumflug, 2016-05-30:
Good. Looks like I'm done. Just moved branch *4bit* to *experimental*, so this type of display is now official. Removed branches *4bit*, *lcd* and *lcd2* as all their contents is on *experimental* now.

[image]

The only thing left is that I lack a bit imagination on what to put on these two lines of text. Perhaps somebody else has and idea and implements something more meaningful. ...

@jgrjgr, are you a mind reader? You write the moment I do, 26 days after the last comment. What a nice correlation :-)

Anyways. For a menu system we need some button input, so I consider this to be a distinct task. Feel free to hack along, writing to the display is now as simple as one can imagine (only beaten by Apple's Cocoa, which needs no code at all :-) ) For example code see `display_greeting()` and `display_clock()` in [display_hd44780.c](https://github.com/Traumflug/Teacup_Firmware/blob/experimental/display_hd44780.c). Code doesn't have to be there, send_P(display_writechar, ...) should work from everywhere. Just don't forget to wrap such code in `#ifdef DISPLAY`.

jgrjgr, 2016-05-30:
lol my missus wishes i was, 

if i remember right most of the hard bits of the code should be done already, i'm pretty sure we had file listing working and even some encoder action as well as a very crude simple menu system running on a rumba board with a 4 line reprap discount smart lcd, it should be in one of the branches , i'll have a poke around with it in a few days,

jgrjgr, 2016-05-30:
on the subject of what to put on the lcd screen, i think really the bare minimum, but also have it configurable for different applications eg for 3d printing say the hotend and bed temps on the first line and the filename on the second

or for cnc usage perhaps x y and z coordinates and have the filename being executed

alfadex, 2016-06-22:
could you also add remaining time?

Traumflug, 2016-06-22:
A firmware can't predict the future, so no, not possible.

Adding [M70](http://reprap.org/wiki/G-code#M70:_Display_message) for displaying arbitrary messages would be trivial, allowing the host or G-code generator to insert related messages.

dcousens, 2016-06-23:
> A firmware can't predict the future

At least,  not very well,  given our limited constraints and environmental factors.

Wurstnase, 2016-06-23:
No, the firmware don't know which gcode will send in the future.

Traumflug, 2016-06-23:
Thinking of it I vaguely remember there are firmwares doing such a prediction? An M-code early in the G-code saying "This print will take 4915 seconds" ? Then we could simply count that number down while receiving additional G-code until it hits zero.

This would require host support, of course. Not sure what these hosts / G-code compilers support.