Traumflug/Teacup_Firmware #216: CoreXY gets confused after homing one axis

phord, 2016-05-25:
After homing an axis and then trying to move to a new position on my new corexy printer, the position targeted is completely incorrect.  I've managed to reproduce this in the simulator using code that I've added recently which simulates endstops (correctly), tracks printer settings from config.h, and converts corexy steps back into a head location.

I don't have a chance to push to a branch here at the moment, but my code and test case is here: https://github.com/phord/Teacup_Firmware/tree/issue-corexy-home

In testcases/homing-corexy.gcode you can see the code which demonstrates the problem:

    G1X10Y10F3000
    G28X
    ; Watch Y after we home X. We do not change Y in the gcode, so Y should not drift.
    G1X10F3000
    ; The printer should report we are at X:10 Y:10. The step counters should match.
    G28Z

The last line (z-homing) is really there to force the simulator to play out the whole movement.  It's not otherwise important to this bug.

At the end of this movement the X and Y position counters should be almost the same.  But what actually happens during the "G1X10" movement is that Y moves a lot and X moves a little.  This seems to be because the code has got confused about where startpoint is, but I haven't verified this deeply.

I will be travelling for a few days and won't get to look at this until next week.  I am leaving this issue here in case anyone else is already looking.

Traumflug, 2016-05-26:
Thanks for the investigation, @phord. As these commits touch almost only simulator code I simply applied them. With the usual whitespace nitpicking, of course :-)

phord, 2016-05-29:
G92 also induces this bug the same way homing does. 
Relative movement is not directly cause this bug, though.

Still investigating.

Traumflug, 2016-05-29:
Are you aware of [dda_new_startpoint()](https://github.com/Traumflug/Teacup_Firmware/blob/master/dda.c#L120)? I'd guess this needs to be more complicated, because axis distance and motor steps are no longer a 1:1 relationship on CoreXY.

phord, 2016-05-31:
Thanks for the tip, @Traumflug.  That's exactly where the problem was.  I thought I checked that, but it was a different function I was looking at.  This is fixed now on experimental.

closed 2016-05-31 by phord

phord, 2016-05-31:
While I'm thinking about this:  It occurs to me that acceleration (and, I assume, lookahead) is calculated against individual axes rather than the motion of the whole head.  But on the corexy movement, the converted axis movements for each stepper motor do not tell the whole story. It is probably more useful to do the math on the physical X and Y motion before converting it to motor steps since the combined action of the two motors results in a movement in the X direction and/or the Y direction, but each motor step does not correlated directly to one axis or the other.  

Delta and Scara probably need similar considerations. Maybe.  Something to think about, but I haven't thought about it enough yet.

triffid, 2016-06-01:
When motors don't directly line up with an axis of motion (everything
except cartesian), it might be necessary to cap acceleration in both head
and actuator space - not sure how feasible this is on an avr8 though!

csdexter, 2016-06-01:
Also known as the ADBI vs. ADAI debate (acceleration/deceleration before/after interpolation).

Traumflug, 2016-06-01:
Strictly physically one needs both. There are parts attached directly to the stepper and parts attached to the extruder carriage. Each of these parts has a mass which needs force to achieve acceleration. Looking at Deltas, the required acceleration force also depends on the current carriage position.

Taking a multitude of acceleration limits into account isn't too complicated: calculate acceleration time for each of them, take the slowest one and calculate back to stepper accelerations from there. Bottlenecks are more in the area of required math and/or a possible plentitude of parameters a user would have to adjust. The latter likely never happens, e.g. tasks like tuning PID is a lot more than what users want to deal with already.

commit b317ba08 (2016-05-31): corexy: Fix new_dda_startpoint for corexy

Add a function axes_um_to_steps to convert from um to steps on all axes
respecting current kinematics setting.

Extend code_stepper_axescode_axes_to_stepper_axes to convert all axes,
including E-axis for consistency.

It seems like axes_um_to_steps could be simplified to something like
"apply_kinematics_axes()" which would just do the transformation math
in-place on some axes[] to move from 'Cartesian' to 'target-kinematics'.
Then the original um_to_steps and delta_um code could remain untouched
since 2014.  But I'm not sure how this will work with sc [... cut]