Traumflug/Teacup_Firmware #219: avrdude parameters on Windows

Traumflug, 2016-06-20:
A Teacup user sees this when trying to upload:
```
"E:\Arduino\hardware\tools\avr\bin\avrdude.exe"
  -C "E:\Arduino\hardware\tools\avr\etc\avrdude.conf"
  -c stk500v2  -b 115200 -p atmega2560 -P COM8
  -U flash:w:"C:\Users\Max Trinks\Desktop\Teacup_Firmware-master\teacup.hex":i
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
```
Obviously the ATmega doesn't receive a reset, but given that uploading with Arduino IDE works, which parameter needs a change here? Sorry for asking, I have neither an Arduino nor Windows for testing.

Wurstnase, 2016-06-20:
My guess is a reset. When you try to upload a binary to the Due over the native port, you need to reset the Due with connecting to it with 1200 baud.

And after a short search I found this:

> The Arduino IDE explicitly toggles the DTR line before uploading (which resets the board) whereas using avrdude from the command line relies on the fact that the standard Linux driver for the FTDI chip sets DTR when you open the serial connection.

And this post: http://forum.arduino.cc/index.php?topic=37586.msg277552#msg277552

Wurstnase, 2016-06-20:
Any special config? Which Arduino Version? I have a Mega here and could test it tomorrow on Win7.

Traumflug, 2016-06-20:
As far as I could find out it's one of these typical cheap Arduino/RAMPS sets.

Vaguely I remember programmer type "stk500v2" doesn't work for every Arduino. If this turns out to be the issue I fear I have to search for a solution to figure out the programmer type automatically.

phord, 2016-06-20:
Or just instruct the user to press reset before uploading.

phord, 2016-07-01:
I haven't tried this on Windows, but I did some investigating on a separate issue this week and found out some details.  The Arduino boards have a capacitor wired from ^RESET to DTR or RTS on the serial line.  This is used to reset the boards and put them in programming mode when you first connect.  Some serial drivers will raise DTR and RTS when you connect to the COM port (windows) or /dev/tty or whatever.  But this is configurable and varies per user system.

avrdude can do this for you automatically, but you have to use the right "programmer" option.

     Wiring boards are supported, utilizing STK500 V2.x protocol, but a simple DTR/RTS toggle is used to set the boards into programming mode.  The programmer type is ``wiring''.

     The Arduino (which is very similar to the STK500 1.x) is supported via its own programmer type specification ``arduino''.

So if Programmer-Type (avrdude -c parameter) is set to `stk500v2`, it does not try to reset the processor with a DTR/RTS toggle.  If it is set instead to `wiring` it uses the stk500v2 protocol, but it precedes the session with a DTR/RTS toggle.  

Similarly, `arduino` is the same as `stk500v1` with an added DTR/RTS toggle beforehand.

I have been fighting this on my Mega2560 RAMPS boardI've been using the last couple of months on a new printer with `-c stk500v2`.  Now that I changed it to `-c wiring` it performs the reset for me automatically.

Problem solved!