Traumflug/Teacup_Firmware #223: M221 extruder override

Traumflug, 2016-06-27:
Let's move the discussion here, I fear these inline comments are difficult to find unless one gets notification emails.

Discussion so far: https://github.com/Traumflug/Teacup_Firmware/commit/257c561e8cfe7a01405120fa0a55c628df7a4079#commitcomment-18014151

As a preliminary result of this discussion I created a second branch, *M221-2*. It moves extruder axis handling over to dda.c, then applies loop unrolling, then the new command. Should work the same way as branch *M221*.

Regarding binary sizes, branch *M221-2* adds 300 bytes, of which 110 bytes are for the introduction of the M221 code; the remaining chunk is for abstraction and unlooping.

Unfortunately I currently have no setup for testing handy, could you please test, @Wurstnase ?

Wurstnase, 2016-06-27:
My first tests were only in the AVR-simulator. I made a simple gcode for it:

```
G91
G1 E5 F500
G1 E-5
G4 P4
M221 S50
G1 E5
G1 E-5
```

The time is the same, while the steps are halved.

phord, 2016-06-27:
This looks like a 4% performance improvement, at least in the "LED ON" times.  I'm not sure what that means, though, with regard to the overall performance.  Can you translate it for me?

Regarding the M221 support, I think the multiplier should be applied against the um variable instead of the steps.  At least in absolute-E mode, this would prevent an accumulation of rounding errors over time which could cause extrusion inconsistencies.

We surely take a little performance hit for applying this extrusion multiplier.  It would make more sense to me if we did not apply it at all when e_multiplier==100.

Finally, it seems that "M221 S0" could be expected to perform a dry-run with no extrusion at all.  But this code does not accept S0 and turns it into S100 instead.  Should we be expected to handle S0 to mean "extruder off"?

phord, 2016-06-27:
Ignore my comment about um vs. steps.  I just occurred to me after I posted it that steps != delta_steps, so the it probably doesn't matter which type gets the adjustment.

Wurstnase, 2016-06-27:
> We surely take a little performance hit for applying this extrusion multiplier. It would make more sense to me if we did not apply it at all when e_multiplier==100.

Yes, I thought the same. But I think, this part is not that time-critical. Also unrolling is not that problem. It is just an idea. I've done this in one commit and should this in two.

The performance winning is not that good as the numbers said. I made a new test of the new formatted code.

Pre-unroll:
LED on time minimum: 3138.44 clock cycles.
LED on time maximum: 5108.8 clock cycles.
LED on time average: 4590.58 clock cycles.

Unroll once:
LED on time minimum: 3117.36 clock cycles.
LED on time maximum: 5087.72 clock cycles.
LED on time average: 4569.5 clock cycles.

Unroll twice (also remove axes_um_to_steps_cartesian)
Adds additional 160 Bytes
LED on occurences: 73.
LED on time minimum: 3016.92 clock cycles.
LED on time maximum: 4987.28 clock cycles.
LED on time average: 4469.06 clock cycles.

Traumflug, 2016-06-27:
Hmm. I'll roll a dice on whether to keep the loops :-)

A thing scratching my head a bit more is whether M221 works the way people expect it. With the current code it's applied to the back of the movement queue while the printer takes movements from the front of the queue. Accordingly any change to extrusion commanded with M221 takes effect only after one round through the queue, which can be some 100 movements. If people are used to set the extrusion multiplier once before starting a print or during filament changes or whatever that's likely fine. Not so sure if they try to change extrusion while the printer is running.

However, getting such changes into effect immediately is tricky, because simply changing steps[E] could mess up the Bresenham algorithm; the axis with most steps could change.

Wurstnase, 2016-06-27:
Yes, this will only applied with new moves. But this is like other firmwares do this. This part of code is mainly to calibrate the printer or to rescue a print. I'm using it rarely, but sometimes. And it is part of many calibration-guides.

[Lets roll the dice.](http://www.roll-dice-online.com/) :+1:

Traumflug, 2016-06-27:
> But this is like other firmwares do this.

Good point! Then Teacup can to this as well, without a shame :-)

... and ACCELERATION_TEMPORAL, in case it ever reaches a usable state, will allow immediate changes.

dcousens, 2016-06-28:
> A thing scratching my head a bit more is whether M221 works the way people expect it. With the current code it's applied to the back of the movement queue while the printer takes movements from the [...]

This is exactly how it works elsewhere,  and how I've often used it when testing different extrusion multipliers at different layer heights mid-print.

:+1:

phord, 2016-07-01:
I think it would be a slight performance improvement if we scaled e_multiplier to 256/100 so the division in dda_create could be optimized to a byte-shift by the compiler (if the compiler does such a thing):

    // Apply extrusion multiplier.
    steps[E] *= target->e_multiplier;
    steps[E] += 128;
    steps[E] /= 256;

I assume that /=256 is significantly faster than /=100, though I haven't tested it.  But maybe someone who is more speed-constrained than I am can comment.

This code will overflow if steps[E] >= 2^24 (16,777,216) steps.  The current code overflows at 2^32/100, or about 43,000,000.  Unlikely in either case, I suppose, but worth considering and noting.

phord, 2016-07-01:
Ok... ignore all that again.  I wrote it before reading @Wurstnase 's comments on my other optimization comments. :-)  Sorry for the noise.  I'll have a useful comment next, I hope.

Traumflug, 2016-07-01:
Huh? This 256-idea is entirely correct, as well as the overflow consideration.

phord, 2016-07-01:
From @Wurstnase earlier:
> Yes, I thought the same. But I think, this part is not that time-critical. 

I just realized I was optimizing the same section again.  So it seems like unwanted noise.  Yes, the idea is valid, but maybe it is not so critical.

But every cycle counts, right?  :-)

Traumflug, 2016-07-01:
> But every cycle counts, right? :-)

Especially if one can save a thousand of them in often repeating code by such a simple change :-)

phord, 2016-07-01:
This feature is timely for me because I am lazily bringing up a new printer while I also work on $dayjob. So I don't have lots of time to tinker with it. Experimenting with build results goes faster if I can change the extruder flow rate with a knob, and I can do it without thinking about it too much. Movement feedrate also affects quality, of course.  So today I also implemented M220 based on this running change.

It was fairly easy to do, but it caused me to reconsider the use of `target` vs. `dda->endpoint` in `dda_create()`.   In the beginning we memcpy them to the same value. But it seems we always say `target` when we read some value from the structure, but we say `dda->endpoint` when we decide to change the data somehow.  But the code just seems to "know" in advance which values might change and which are safe to read from `target` since they will remain unchanged.  This pre-knowledge is unwelcoming to new developers and is unsafe for old ones.  

In the process of verifying and validating some of these assumptions, I made `target` const and switched over all the subsequent access to use dda->endpoint instead.  I didn't think about it much outside the `dda_create` function, though, and it's possible I missed some important implications, like _who saves the feedrate during homing and do they care if M220 modified it in the middle?_

So maybe my change is over-complicated.  But I'll split it up to help make sense of it.

phord, 2016-07-01:
I implemented the 256-scaled percentage value and pushed it to M221-2 at 3dec853d2c7a9ceb4b1397f495d378025bb188bd.  I checked the assembler output and verified it does remove a call to udiv and replaces it with 3 byte moves.  Score!

I also implemented M220 (feedrate override percentage) and pushed it to [M220](https://github.com/Traumflug/Teacup_Firmware/compare/M221-2...M220).

Wurstnase, 2016-07-01:
@phord Great! I was looking in that part alot last days, because when we have M221 we need a M220 also. But I had no idea. With little luck I will start tomorrow with changing the complete electronic from the Due to my STM-Solution and can test all this new features live. 👍

Traumflug, 2016-07-02:
Dices fell for the smaller, slower version, still I made a comment for the faster variant to not loose this knowledge. Branch *M221-2* is now ready for *experimental*, let us know how the live test worked :-)

phord, 2016-07-02:
It looks sane, but I can't test it this weekend.  I tested the previous version yesterday and it worked quite nicely.  :-)

Wurstnase, 2016-07-07:
Currently I found no time to assemble it on my printer. Anyhow, I tested it on my desk. It works as intended! M220 and M221. :+1:

Traumflug, 2016-07-09:
Excellent. Moved branch *M221-2* to *experimental* and then removed branches *M221-2* and *M221*. Thanks a lot for your contributions.

Traumflug, 2016-07-09:
Same for branch *M220*. Moved to *experimental*, then deleted.

This time after adding a pair of regression tests for ACCELERATION_TEMPORAL and ACCELERATION_REPRAP.

Looks like we're done here :-)

Wurstnase, 2016-12-19:
Looks like we can close this issue.

closed 2016-12-19 by Wurstnase