Traumflug/Teacup_Firmware #224: X is not critical anymore?

Wurstnase, 2016-06-30:
https://github.com/Traumflug/Teacup_Firmware/blob/master/dda.c#L395
I guess it's gone here: https://github.com/Traumflug/Teacup_Firmware/commit/3988e474

phord, 2016-07-01:
I think your assessment is correct.  I've been ignoring this comment a lot lately as I experiment with different acceleration algorithms.

Traumflug, 2016-07-01:
The "critical" thing here is like this: X has an acceleration of 1000, Y has an acceleration of 50. Now imagine a movement where X does slightly more steps than Y, so X becomes the "fast" axis. The bot will then try to accelerate with 1000, almost 20 times more than on the almost fast axis (Y) allowed.

The right way would be to calculate acceleration time for each participating axis, take the slowest one and calculate back to how fast the axis with the most steps can accelerate to achieve this (equal or slower) acceleration time.

This isn't an entirely theoretical question, movements with X and Z combined expose this problem already on commonly used printers (and the fact there is no distinct acceleration for the Z axis).

phord, 2016-07-01:
Originally this comment referred to some line (26fb18d1) which calculated the initial `c` by considering the acceleration only for the X-axis, regardless of which axis was the "fast" one. This limitation was long since removed. 

Since ACCELERATION is only given once and not per-axis, you can be assured that the axis which takes longest to accelerate to its top speed -- the limiting axis -- is also the axis which will be going the fastest during this movement.

Traumflug, 2016-07-02:
Good. Forwarded the commit to *experimental* and removed the branch. Thank you very much :-)

closed 2016-07-02 by Traumflug