Traumflug/Teacup_Firmware #227: Welcome aboard, Daniel!

Traumflug, 2016-06-30:
The rules here are simple: keep fingers off branches *master* and *experimental* for now, but create as many own branches you feel need for. In Git parlance they're topic branches, so their commits can be refined in place and follow recent development by rebasing until it's reasonable sure the goal of the change/feature is achieved.

Yes, I'm aware you might be waiting for a Gen7 with 5th extruder, so I'll dedicate today to get this finished. Until then I have only a picture:

[image]

The pins available on these two SPI connectors can be used to drive a second extruder. The one pin for SPI1 is the Enable pin, which could be taken from the other Enable pins, too. Freeing SPI1 doesn't free the SPI bus, though, most pins are shared between SPI1 and SPI2.

dcousens, 2016-06-30:
Thanks @Traumflug, as discussed I'm intending to follow up the above hardware with the appropriate working firmware for a 5th extruder (aka motor + heater + thermistor).

Right now,  I'm hoping to get my head around the structure of the project and its inner workings so that I'm context aware before I need to start the above.

dcousens, 2016-06-30:
@Traumflug I've noticed in #204 that the major issue with that change was the large amount of unresolved conflicts in rebasing other branches.

Is there a list of "officially" supported branches that are to be kept up to date such that they must be rebased?

Traumflug, 2016-06-30:
Basically all branches have some justification. Each of them contains an idea, some improvement, some code which should be evaluated and be either discarded after careful consideration or be adjusted to enter mainline. "Stale" branches aren't stale because they contain nonsense, but because nobody reworked them to enter mainline.

A good example is branch *scara*, which adds support for Scara printers. But nobody tested this, no support in Configtool, it might break other kinematics types, some reverting commits which should be squashed, stuff like this. On how to deal with them see issue #81.

For the second extruder thing, branches *lumentino* and *multi-extruder* add some support. Certainly old enough to count as "stale", still valuable code. Task would be now to look what's the distinction between them and how they can be adjusted to fit modern knowledge and user expectations. Most of the commits on these branches should simply go away during step-rebasing, as the urgent parts were picked to *experimental*/*master* already.

dcousens, 2016-07-03:
I might also have a stab at `G2/G3`

Traumflug, 2016-07-03:
There is a branch *arc-support* which claims to support G2/G3. Cleanup opportunities I see at a first glance:

- Should be wrapped into a #define, because it's likely a big chunk of binary size.
- motion_control.c/.h needs cleanup and a proper name.
- There is some "E endstop" ... should go into a separate commit, be allowed to be absent and evaluated for usefulness.

As far as I can see, this implements G2/G3 by segmentation. Not ideal, but certainly better than nothing.

Traumflug, 2016-07-09:
After messing a few days with OpenJDK 9 (can't run Visolate), then OpenJDK 8 (can run Visolate), then building a JAR with a Makefile and doing a number of non-software duties, something appeared at last:

[image]

Two of these four are for you, @dcousens :-)

dcousens, 2016-07-11:
Getting excited @Traumflug :)

dcousens, 2016-07-30:
Received!
Will keep this thread up to date with my progress.

dcousens, 2016-09-27:
Status update:

I broke everything and got bogged down with several projects for work.
I intend to follow through on my intentions above around the Christmas period.

dcousens, 2017-01-03:
I have not forgotten,  I'm currently cleaning out my garage to get back to this!