Traumflug/Teacup_Firmware #228: Rework acceleration ramps calculations

Wurstnase, 2016-07-05:
I want to start with the first commit.

`acc_ramp_len()` and `ACCELERATRE_RAMP_LEN()` could need a small rework.
With https://github.com/Traumflug/Teacup_Firmware/commit/705e4563db075 we find, that X is not critical normally, but it is used somewhere else. In this step I saw, that `dda->fast_spm` is only used once.

So my idea is to remove this part and let the precompiler do some stuff instead. https://github.com/Traumflug/Teacup_Firmware/commit/8530049b9a

Wurstnase, 2016-07-05:
With the last 4 commits we save 110 bytes flash and 32 bytes ram.
https://github.com/Traumflug/Teacup_Firmware/commits/acc_ramp_len

phord, 2016-07-05:
Nice work.

This is curious to me:  `ACCELERATE_RAMP_LEN` is still there and untouched by this series. Replacing the three calls to it with `acc_ramp_len` and using some specific (but possibly incorrect) axis lookup in each case results in _no change_ in code size.  

    -     // TODO: calculate the steps from the fastest axis and not from X.
    -    prev_F_in_steps = ACCELERATE_RAMP_LEN(prev_F);
    -    this_F_in_steps = ACCELERATE_RAMP_LEN(this_F);
    -    crossF_in_steps = ACCELERATE_RAMP_LEN(crossF);
    +    prev_F_in_steps = acc_ramp_len(prev_F, current->fast_axis);
    +    this_F_in_steps = acc_ramp_len(this_F, current->fast_axis);
    +    crossF_in_steps = acc_ramp_len(crossF, current->fast_axis);

For correctness, fast_axis should be looked up in the ATOMIC section, but here it is not just for this example.  Also, I'm not sure of the math here, so can't decide if this is even correct.

Wurstnase, 2016-07-05:
Exactly. This part needs also a change. I'm also unsure how this works and want to start to ask someone.

Wurstnase, 2016-07-11:
Looks like independent axis works. Maybe there is someone who can test it?!? Unfortunately I have no idea how. My test with 4 motors on my desktop looks like OK.

https://github.com/Traumflug/Teacup_Firmware/commit/9919f79273a

Wurstnase, 2016-07-12:
Ok, it's done. My simulations looks quit good. https://github.com/Traumflug/Teacup_Firmware/commit/9daa942f6e

Also the speed, acceleration etc. are the same on different axis-steps-per-m. https://github.com/Traumflug/Teacup_Firmware/commit/b75bb5d3
One axis looks like a bit noisy in gtkwave, but this comes from the script.

[image]
[image]

Traumflug, 2016-07-12:
If you man the upper picture, 730 ms to 890 ms, with "a bit noisy", this is the Bresenham algorithm at work. Slower axes can step in multiples of the fast axis, only, so slower axes usually switch between two speeds all the time.

What is the second picture? Except for this spike at 822 ms it looks quite good!

Wurstnase, 2016-07-13:
Second one is current experimental with equal steps/m on X and Y.

Wurstnase, 2016-07-13:
const PROGMEM for the divisor is faster and needs same space.

On the STM32
https://github.com/Traumflug/Teacup_Firmware/commit/9daa942f6e = 10 cycles
https://github.com/Traumflug/Teacup_Firmware/commit/05b0c7be34bb1c0 = 7 cycles

btw. I test the previous inline code. That was ~14 to 28 ticks.

Traumflug, 2016-08-09:
Experience report from the RepRap forum:

> hi, i try this [branch acc_ramp_len] but now feedrate is verrry slow. it is like maximun feedrate is adjusteed very low and i cannot try print
> 
> edit: i try (dda.c: take same calculation for move duration in ramping from temporal ) there is normal speed fedrate , but it is slow feddrate for homing axis

Wurstnase, 2016-08-09:
I'm simulating some HEAD~x and find an issue. But this should not be a problem for ACCELERATION_RAMPING in the last stand.

But regarding to [this comment](https://github.com/Traumflug/Teacup_Firmware/blob/master/dda.c#L845): `// A "-=" would overflow earlier.`
I changed the code from `A /=` to `A = A /` which solves an issue. (This code is only active for ACC_REPRAP)

I don't find any relation between homing and that new code. Homing is only normal moving with endstop active!?

Wurstnase, 2016-08-09:
I'm getting closer...

Wurstnase, 2016-08-09:
Ok, what is the difference between inline in a .h-file and not inline-ing in a .c-file?

I take the code from dda_math.c:
```
uint32_t acc_ramp_len(uint32_t feedrate, uint8_t fast_axis) {
  static const axes_uint32_t PROGMEM acc_ramp_div_P = {
    (uint32_t)(((double)7200000 * ACCELERATION) / STEPS_PER_M_X),
    (uint32_t)(((double)7200000 * ACCELERATION) / STEPS_PER_M_Y),
    (uint32_t)(((double)7200000 * ACCELERATION) / STEPS_PER_M_Z),
    (uint32_t)(((double)7200000 * ACCELERATION) / STEPS_PER_M_E)
  };
  
  return (feedrate * feedrate) / acc_ramp_div_P[fast_axis];
}
```
to the header as:
```
static uint32_t acc_ramp_len(uint32_t, uint8_t) __attribute__ ((always_inline));
inline uint32_t acc_ramp_len(uint32_t feedrate, uint8_t fast_axis) {
...
}
```
and everthing is perfect in SimulAVR...

This code will replace the current ACCELERATION_RAMP_LEN-macro.

triffid, 2016-08-10:
With regular function in C, the function has an address. Wherever it is
used, the compiler sets up the stack and jumps to the function's address,
with address of next instruction as the return address. This means that the
function can be used any number of times with very little increase in code
size, however it also requires setting up the stack and jumping whenever
the function is used, which takes a little time.

With inline in H, the function has no address. Instead, the compiler
inserts its body verbatim wherever the function is used. This increases
code size by (size of function) * (number of times function is used)
however it also gives faster code execution and can allow greater
optimisation.

An inline function is similar to a macro, except it's handled by the
compiler rather than the preprocessor and thus type-checking and suchforth
occurs.

The best candidate for an inline function is a function that must be fast,
and is either very small or used infrequently.

Wurstnase, 2016-08-10:
@triffid Thanks for explanation. I thought this was the issue. But in the end it seems to be a missing pgm_read_dword for the const array.

Also the move_duration needs some more work. I have a local copy of it and will do some work later.

The code should now work. Only for different steps per axis is not accurate currently. I will spend later more time for this.

Wurstnase, 2016-09-15:
Any idea what this magic '2' does?
[First commit](https://github.com/Traumflug/Teacup_Firmware/commit/1e6c74096eb)
[Line 57 of first commit](https://github.com/Traumflug/Teacup_Firmware/blob/1e6c74096ebce70884b5620e7142afeac7d7e859/mendel/dda.c#L57)
[Current code](https://github.com/Traumflug/Teacup_Firmware/blob/master/dda.c#L289)

```
if (distance < 2)
  distance = delta_um[E];
```

Traumflug, 2016-09-15:
Looks like a `(distance == 0)` with some allowance for inaccuracies of approx_distance().

Wurstnase, 2016-09-15:
Ah, yes. It's 2um ~ 0.

phord, 2016-09-15:
I think it could be written more clearly.  But also it seems like it could be a bug.  What if distance==1 and delta_um[E]==0?  Maybe it should say 

    if (distance < delta_um[E])

instead.  But maybe git-blame can tell us what was meant.  I can't check right now.  I can only make noise.  :-)

Wurstnase, 2016-09-15:
@phord yes I think something in that direction also. Think about distance is 3 and delta_um[E] is huge.

Traumflug, 2016-09-15:
Keep in mind that the speed given by F counts for X,Y, Z, A, B, C, I, J, K only, but not for the extruder axes. Having an E-only movement is the exception.

Also that calculated distance is currently used for speed calculation, only. Having a small X/Y movement and a large E movement should properly limit X/Y speed to not exceed E's limit.

Wurstnase, 2016-09-15:
This makes sense. So we always try to get perfect speed on the axis and not in the extruder. This is good.

phord, 2016-09-16:
Sometimes I see a bug when I abort a print and restart it, Teacup will try to rewind the extruder to position 0.000.  This is bad enough, but sometimes it seems to run the extruder stepper at an insane speed, so eventually the stepper is just whining but not moving at all. It is probably caused by some missing or unneeded G-code in my startup script.  I don't see it every time, so it's probably in some slicer I don't use often.

But seeing this code and @Traumflug 's comments about E-speeds reminds me of it.  I wonder if it's related, or if it is only caused by me having insane MAX speeds for my stepper in my config.  Something to look at when I get my printer back out.