Traumflug/Teacup_Firmware #229: No deceleration-ramp at hi step rates

Wurstnase, 2016-07-10:
Current setup 1280 steps/mm or 640 steps/mm.
Acceleration is independent. Same issues at 500 or 1500 mm/s². Jerk has also no influence.

With 1280 steps/mm and a feedrate of 14851 mm/min everything is ok. Smooth acceleration, smooth deceleration. At 14852 I have a smooth acceleration but no deceleration anymore.

Also at 640 steps/mm and a feedrate of 29703 mm/min. No more deceleration.

Wurstnase, 2016-07-10:
Hmmm... Don't now why this makes the different. Low Jerk helps. But it is not the Jerk. I can hear the motor starting decelerate at one steprate X. On X+1 I can't hear any deceleration anymore.

Also I changed now the timer. Maybe it has also something to do with this comment?
https://github.com/Traumflug/Teacup_Firmware/blob/master/timer-arm.c#L156-L161
```
Flipside of this coin is, schedules issued at an arbitrary time can result in drastically wrong delays.
  See also discussion of parameter check_short and the return value.
```

I reset the Counter to 0 and set the compare register to `delay`.

stm32f411
TIM5->CNT = 0;
TIM5->CCR1 = delay;

LPC1114
LPC_TMR32B0->TC = 0;
LPC_TMR32B0->MR0 = delay;

With that, it looks like I can reach 1,3MHz-steprates. (without osci)

Traumflug, 2016-07-10:
> Flipside of this coin is, schedules issued at an arbitrary time can result in drastically wrong delays.

This "arbitrary time" means the first step after a longer time with no step at all. For example the first step of an entire print. In such a case the timer counter is of unknown value and simply adding the step time can cause a wrap-around of the timer ... which can be up to 90 seconds (2^32 / F_CPU). On AVR with the 16-bit timer it's just 3 milliseconds.

> I reset the Counter to 0 and set the compare register to delay.

Then you add the time needed for processing the interrupt to the step delay. This might be not noticeable, because this processing time happens in the highest priority interrupt and is therefore pretty constant.

To deal with this situation, there's another comment a few lines later:

    If there is no previous step, TC and MR0 should have been reset to zero
    by calling timer_reset() shortly before we arrive here.

This call to timer_reset() happens in dda_queue.c, line 139 ... right after adding a first move into an empty movement queue.

To sum up: these drastically wrong delays can't happen in the middle of a movement.

Regarding the issue with missing deceleration it's likely an overflow in one of the ramping calculations. I'd try with LOOKAHEAD disabled to make the code involved smaller. Acceleration and deceleration should still work then.

Wurstnase, 2016-07-11:
Ok, I will check this.

While viewing the code I found this line. https://github.com/Traumflug/Teacup_Firmware/blob/experimental/dda_maths.c#L273
`(((uint32_t)7200000UL * ACCELERATION) / steps_per_m)`
This part will overflow at `ACCELERATION > 596`?!?

Wurstnase, 2016-07-11:
The (uint32_t) has moved in this commit: https://github.com/Traumflug/Teacup_Firmware/commit/6fae5a8b7cb77

Traumflug, 2016-07-11:
It moved from being a float to being an int (note the different function used), which looks more crucial to me.

The commit you found doesn't exactly look well considered. No idea what I had in mind back then.

Traumflug, 2016-07-11:
Scratch that. ACCELERATE_RAMP_LEN_SPM and ACCELERATE_RAMP_LEN are two distinct macros. Apparently I'm as confused as back then :-/

Regarding a solution: at the first glance it looks like a good idea to mix the recent enhancement in branch *acc_ramp_len* with the older, slightly modified macro:

      #define ACCELERATE_RAMP_LEN(speed, spm) (((speed) * (speed)) / \
                (uint32_t)((7200000.0f * ACCELERATION) / (float)spm))

Perhaps it works using an uint64_t instead of a float without bloating code, too, giving better accuracy. Like:

      #define ACCELERATE_RAMP_LEN(speed, spm) (((speed) * (speed)) / \
                                               (((uint64_t)7200000.0 * ACCELERATION) / spm))

Wurstnase, 2016-07-11:
With https://github.com/Traumflug/Teacup_Firmware/commit/2c3c111ba40ea9e2181729c5f45320226b964da0 we could change the function (we could remove this macro) to:
`divisor = (7200000UL * ACCELERATION / STEPS_PER_M_X);`
The preprocessor should do the right math until this is <32bit.

Traumflug, 2016-07-11:
I'm pretty sure such functions are compiled first, then optimized away. Preprocessor function evaluation happens only in #if instructions. You can check by looking at the *build/dda.i* file after a compile, that's the preprocessor output. In the .i file macros are expanded, but variable names still kept, so you can search for 'divisor'.

Inserting a sersendf_P() to see it at runtime works, too, of course.

That said, a good optimizer would recognize common parts of math terms, so commits like these: https://github.com/Traumflug/Teacup_Firmware/commit/0424fac62d11b0332547d64dce291c6dd3156574 https://github.com/Traumflug/Teacup_Firmware/commit/56bf1e15ad932203a4541ff5d69b7f150ef888a2 https://github.com/Traumflug/Teacup_Firmware/commit/46caa0bd49199d6572ff99177403f2c993dd044d whould be redundant. Apparently the optimizer isn't that good.

Wurstnase, 2016-07-11:
Checked:
```
      divisor = (7200000 * ACCELERATION / STEPS_PER_M_X); -> Overflow
      divisor = (7200000UL * ACCELERATION) / (STEPS_PER_M_Y); -> OK
      divisor = (7200000UL * ACCELERATION / STEPS_PER_M_Z); -> OK
      divisor = (7200000 * ACCELERATION) / (STEPS_PER_M_E); -> Overflow
```

Wurstnase, 2016-07-11:
Ok... And I found my mistake... A missing break...

```
uint32_t acc_ramp_len(uint32_t feedrate, uint8_t fast_axis) {
  uint32_t divisor;
  switch (fast_axis) {
    default: // default is also the X-Axis
    case X:
      divisor = (7200000UL * ACCELERATION / STEPS_PER_M_X);
      break;
    case Y:
      divisor = (7200000UL * ACCELERATION / STEPS_PER_M_Y);
      break;
    case Z:
      divisor = (7200000UL * ACCELERATION / STEPS_PER_M_Z);
      break;
    case E:
      divisor = (7200000UL * ACCELERATION / STEPS_PER_M_E);
      break;
  }
  return (feedrate * feedrate) / divisor;
}
```

Wurstnase, 2016-07-12:
Hmmm... Any of these cases will overflow compiled on my laptop.
> divisor = (7200000 * ACCELERATION / STEPS_PER_M_X); -> Overflow
> divisor = (7200000UL * ACCELERATION) / (STEPS_PER_M_Y); -> OK
> divisor = (7200000UL * ACCELERATION / STEPS_PER_M_Z); -> OK
> divisor = (7200000 * ACCELERATION) / (STEPS_PER_M_E); -> Overflow

Traumflug, 2016-07-12:
TBH, no surprise, because 2^32 / 7200000 = 596.

Wurstnase, 2016-07-12:
Ok, it's done. My simulations looks quit good. https://github.com/Traumflug/Teacup_Firmware/commit/9daa942f6e

Also the speed, acceleration etc. are the same on different axis-steps-per-m. https://github.com/Traumflug/Teacup_Firmware/commit/b75bb5d3
One axis looks like a bit noisy in gtkwave, but this comes from the script.

[image]
[image]

Wurstnase, 2016-07-18:
Ok, the STM32 needs ~190 cycles for dda_step(). So we should not be faster. This is at 100MHz, 526kHz step rate.

While testing I found another issue.

Setup is 1280000 steps/m with 500 acc. No lookup.

In dda_clock() I add a `serial_writechar('>');` in `if (move_step_no < dda->rampup_steps)`
and `serial_writechar('<');` in `else if (move_step_no >= dda->rampdown_steps)`.
I add also `if (! recalc_speed) serial_writechar('x');`

I reduce the TICK_TIME to 10MS.

G1 y250 F19600:
total_steps:320000 distance:250000 move_duration:5062500
rampup_steps:136600
rampdown_steps:183400
total_steps:320000
`>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>xxxxxxxxxxxxxxxxxxx<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<`

G1 y250 F19700:
total_steps:320000 distance:250000 move_duration:5062500
rampup_steps:138012
rampdown_steps:181988 
total_steps:320000
`>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>`

As you can see, the acceleration works flawless until it reaches the saddle of movement. Then anything stops. Any timer work perfect. No overflow or waiting. I can immediately start a new move.

Any ideas where I can search?

Wurstnase, 2016-07-18:
A small update. This happens not every time. Just made a new test.
10 times `G1 y250 F19600`. 4 are ok. 6 fails after acceleration stage.

Wurstnase, 2016-07-18:
Next step.
That we exactly hit the point where accelerations end could be come from inaccuracies of int_inv_sqrt(). I changed it to `/ (2 * int_sqrt())` and the motor stops a bit later.

Interesting point, when the motor stops while decelerating, in dda_clock it don't go in the `else if (move_step_no >= dda->rampdown_steps)`. move_step_no is then around 199800 of 320000 total and rampdown starts with 186160.

Traumflug, 2016-07-18:
Looks like you're in the right track.

Wurstnase, 2016-07-18:
Well, it could be the sqrt itself, or the complete dda_clock().

When the `dda->c` goes below 270, it looks like, dda_clock() will not continue and is long time waiting for dda_step(). In that case dda_clock() goes to the sqrt-calculation. But the result is waiting while the stepper will finish its move.

In the most efficient algorithm it need ~ 50 cycles to calculate the `c0 / 2*sqrt(move_n)`. With interrupts sometimes 300 cycles. In a straight move `G1 y600` we have at speed below 300mm/s 50 cycles min, 300 cycles max and a mean of 100. Slightly faster and the max cycles will grow to 530. Mean is still close to 100 an then starts to increase. 

When the mean reaches >135 the dda_clock() has no time anymore to get finished.

closed 2016-07-19 by Wurstnase

Wurstnase, 2016-07-22:
Just for the completeness. I made a fast an dirty double stepping in dda_step(). Now I can reach the limit of it at 190 cycles, or in other words 565,000 steps/s.