Traumflug/Teacup_Firmware #231: Ramps 1.4 compilation error

jcca, 2016-07-20:
`make -f Makefile-AVR

CC build/mendel.o
In file included from mendel.c:46:0:
spi.h: In function 'spi_select_sd':
spi.h:33:17: error: 'DIO53_WPORT' undeclared (first use in this function)
WRITE(SD_CARD_SELECT_PIN, 0);
^
spi.h:33:17: note: each undeclared identifier is reported only once for each function it appears in
spi.h:33:38: error: 'DIO53_PIN' undeclared (first use in this function)
spi.h: In function 'spi_deselect_sd':
spi.h:38:17: error: 'DIO53_WPORT' undeclared (first use in this function)
WRITE(SD_CARD_SELECT_PIN, 1);
^
spi.h:38:38: error: 'DIO53_PIN' undeclared (first use in this function)
make: *** [Makefile-common:89: build/mendel.o] Error 1`

Wurstnase, 2016-07-20:
Do you select the right processor your makefile? Looks like you try to
compile a smaller one which hasn't a pin 53.

jcca, 2016-07-20:
Yes, I don't understand why don't work with "make -f Makefile-AVR" but when I do "Build->Build   and Build->Upload" it works..

now I am very happy making a pcb, Thanks teacup team.

closed 2016-07-20 by jcca

Wurstnase, 2016-07-20:
The configtool makes this for you automatically. When you want to compile
this via makefile you need to change the mpu in the first lines in
Makefile-AVR to match your system.

triffid, 2016-07-20:
This is what configtool is for.. Default processor in makefile is the '328
which doesn't have pin 53