Traumflug/Teacup_Firmware #234: ESP8266 native port

vinouz, 2016-08-15:
Hi Traumflug, 
I'd be interested in adapting teacup to the ESP8266, using its Arduino board definition (https://github.com/esp8266/Arduino)

Where will I have to start ? What areas need attention ? I guess pin definition files, and timer functions ?

Thanks, 

Vincent.

Traumflug, 2016-08-15:
Nice idea!

First, Teacup is entirely independent from anything "Arduino". In fact, recent versions of Arduino IDE are currently not supported as build platform, because they changed their build algorithm again, messing stuff for Teacup up. Teacup uses gcc (variant depending on the chip) and avr-libc (AVR) / CMSIS (ARM).

First steps would be to get the basics running: building a working binary using using the Makefile, with some communications to the outer world. Which kind of communications doesn't matter. That's the toughest part. With communications working one can make visible what's happening, so things get easier from there on. It's fine to borrow/copy code from other projects, of course.

There's a tag *arm-port-starts-here* in the repository; the __following__ commits show how to port to an ARM platform step by step. There's also branch [*arm-stm32f411-port*](https://github.com/Traumflug/Teacup_Firmware/commits/arm-stm32f411-port) which does exactly this, also step by step. This should give you an idea on how to do the port.

Traumflug, 2016-08-15:
P.S.: looking at this [Hackaday article](http://hackaday.com/2014/10/25/an-sdk-for-the-esp8266-wifi-chip/) I wonder a bit on how you want to get a printer running with these 8 pins. If it's just about using the ESP8266 as WiFi-Serial bridge, I'd guess there is no porting work required at all. Hooking up this tiny board to the serial line of a RepRap controller should just work.

phord, 2016-08-15:
The [Digistump Oak](http://digistump.com/wiki/oak/tutorials/pinout) supports 11 digital and 1 analog pins.  [Huzzah 8266 by Adafruit](https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/pinouts) has 9 digital and one analog.  The difference seems to be that the Huzzah reserves two GPIO for serial RX & TX.

Teacup minimally requires 9 digital and 1 analog pins, I think.  But I don't know if anyone runs it this way. 

8 digital Pins: For each axis, X, Y, Z and E, you must have STEP and DIRECTION

1 digital pin: Extruder heater output
1 analog pin: extruder temperature sensor (might be digital depending on sensor)

So, with no heated bed, no endstop triggers, no SDCard, no LCD output, no hardware flow control, no PS_ON, etc., I think this is feasible.

You can even do away with RX & TX, freeing up the two UART pins for something else, if you rely on network communications with the ESP8266 for all gcode interactions, assuming step interrupts do not get confused over this.

Probably a good place to start, in addition to the porting commits Traumflug already mentioned, is to look at the difference between the ARM and AVR code.  It is mostly isolated in separate targeted source modules:

    Teacup_Firmware $ ls *-avr.c
    analog-avr.c  cpu-avr.c  delay-avr.c  heater-avr.c  serial-avr.c  timer-avr.c
    Teacup_Firmware $ ls *-arm.c      
    analog-arm.c  cpu-arm.c  delay-arm.c  heater-arm.c  serial-arm.c  timer-arm.c

You may also want to look at Makefile-ARM, Makefile-AVR and Makefile-common.

Traumflug, 2016-08-15:
> The Digistump Oak supports 11 digital and 1 analog pins.

Yes, such a board makes much more sense.

One could even get away with one common Dir pin if the Dir pin is switched just before each trigger of the Step pin. Not sure how reliable this strategy is for stepping in the kHz range, though.

Wurstnase, 2016-08-15:
Maybe the successor [esp32](http://www.espressif.com/products/hardware/esp32/overview)?

vinouz, 2016-08-16:
The lolin v3 board sports 6 more pins : 2 gpio and spi/i2c or sdio. That should begin to make an interesting platform. Good I have one ;-).

Thanks, I'll get a look into that ;)

http://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/

Traumflug, 2016-09-06:
The new ESP32 is apparently available now and comes with 33 GPIO pins:

http://www.pighixxx.com/test/2015/12/esp32-pinout/ 
http://hackaday.com/2016/09/05/new-part-day-the-esp32-has-been-released/

ryannining, 2017-06-09:
For ESP8266, i have try the NotGRBL, small CNC firmware for esp8266, and i have success using I2C IO expander to control 3 motor. Using just 2 pin on ESP01. I am sure using shift register IC is better, we can transform just 3PIN into so many digital pinout. And use other pin as Input (3 endstop, 2 temp) its enough.

ESP8266 can be overclock to 160Mhz, and its very cheap.

Traumflug, 2017-06-09:
I think the situation here is similar to that of using V-USB directly on the movement controller chip: it's easy to do this implementation, replacing the serial interface, but USB comes with tight timing restrictions. Accordingly, V-USB occupies the highest level interrupt and stepper motor timing goes down the toilet with heavy USB traffic.

However, USB-equipped ATmegas apparently do work (Teacup has an implementation, there are success reports), because (my guess) these hard timing requirements get handled in dedicated hardware.

As you have such an ESP8266 running already, could you try to find the limits? Like running steppers very fast (50'000 steps/second or more) for longer periods, then also establishing heavy WiFi traffic? If steppers continue to run smoothly, there's a good chance to do a usable(!) port of Teacup.