Traumflug/Teacup_Firmware #235: ConfigTool startup is slow

phord, 2016-08-17:
Recently ConfigTool has been very slow for me on Ubuntu Linux.
When I run the app there is a 15 second wait before the window is
first displayed.  I bisected the problem and found it was tied to
the number of pins in `pinNames`, and ultimately that it was caused
by a slow initializer in wx.Choice() when the choices are loaded when
the widget is created.  Patch coming...

Traumflug, 2016-10-26:
Thanks for the code simplification.

As I didn't see such a dramatic change here I investigated it a bit and noted the results at the bottom of the commit message. I t "feels" a bit faster now, but barely measurable (some 0.5 seconds). Perhaps it's also a matter of compilation, my Ubuntu 16.04 compiles most python code automatically. .pyc files appear by just running configtool.py for all files in configtool/. configtool.py it's self stays uncompiled for some reason.

Anyways, commit moved to *experimental* and topic branch removed. Thanks again!

jbernardis, 2016-10-27:
Python only "compiles" files when they are imported.  Since the main file
is executed directly,  it is never automatically compiled.

commit 6b6b1da8 (2016-08-17): Speed up ConfigTool startup with wx-tricks Fixes #235

Recently ConfigTool has been very slow for me on Ubuntu Linux.
When I run the app there is a 15 second wait before the window is
first displayed.  I bisected the problem and found it was tied to
the number of pins in `pinNames`, and ultimately that it was caused
by a slow initializer in wx.Choice() when the choices are loaded when
the widget is created.  For some reason, moving the load after the
widget is created is significantly faster.  This change reduces my
startup time to just under 4 seconds.

Further speedup could be had by using l [... cut]