Traumflug/Teacup_Firmware #239: Wurstnases branches

Wurstnase, 2016-09-07:
This will be an overview of my branches. This should made the work easier for @Traumflug to see which could be merged/rebased on experimental or so.

I will edit this very first post if anything changed.

[keys_and_display_menu](https://github.com/Traumflug/Teacup_Firmware/tree/keys_and_display_menu)
This will introduce a menu with keys for a display. We have a simple state machine to handle any case we want. Code needs currently some rework.

[fix_ssd1306](https://github.com/Traumflug/Teacup_Firmware/commits/fix_ssd1306)
Should be ready to merge (configtool?!?)
Add Adafruit 128x32 display for i2c
Fix a missing progmem
Add a Teacup-logo and a python script to port a xbm
Add font8x8

[cleaning_dda_lookahead](https://github.com/Traumflug/Teacup_Firmware/tree/cleaning_dda_lookahead)
Traumflug is going to merge this currently (2016/09)

[acc_ramp_len](https://github.com/Traumflug/Teacup_Firmware/tree/acc_ramp_len)
Most code should work. I will take a last look into it.
Remove some redundant code, increase performance, saves a lot of space.

[arm-stm32f411-port](https://github.com/Traumflug/Teacup_Firmware/tree/arm-stm32f411-port)
My little baby. Currently one minor not critical bug in serial (solved on my desktop).
Next step: implementing SPI and maybe I2C.

[acc_temporal](https://github.com/Traumflug/Teacup_Firmware/commits/acc_temporal)
Little bit messy. Just a first idea an needs more work.

[ramp_no_lookup](https://github.com/Traumflug/Teacup_Firmware/tree/ramp_no_lookup)
Solved a bug when using ACC-ramping without lookahead (some missing ifdefs)
Ready to merge, no huge code change

[acc_temporal_hifreq](https://github.com/Traumflug/Teacup_Firmware/tree/acc_temporal_hifreq)
Solves an overflow for frequencies > 72MHz
Ready to merge, no huge code change, no functional change

[inv_sqrt_overflow](https://github.com/Traumflug/Teacup_Firmware/tree/inv_sqrt_overflow)
Needs rework

[pid_autotune](https://github.com/Traumflug/Teacup_Firmware/tree/pid_autotune)
Needs minor rework and a new tab in configtool, but most work is done
Adds a PID-autotune: When active, heaters can only be tuned. There you get the 'autotune'-values. With that you can calculate easily good values for PID.

[temp_prediction](https://github.com/Traumflug/Teacup_Firmware/tree/temp_prediction)
An attempt for a temp prediction instead of PID
I'm not sure if this will work good enough. 14.2 fixpoint could be not accurate enough for good heater control.

[pullups_for_endstops](https://github.com/Traumflug/Teacup_Firmware/tree/pullups_for_endstops)
Different pullup-settings for each endstop

[issue_204](https://github.com/Traumflug/Teacup_Firmware/tree/issue_204)
formating stuff for spaces/tab-issue. Could be deleted.

[arm_arduino_due](https://github.com/Traumflug/Teacup_Firmware/tree/arm_arduino_due)
I've got this never to work. We could delete it or let it. Anyhow, I will not investigate any time into it anymore.

Wurstnase, 2016-09-10:
add progmem_to_flash
update arm-stm32f411-port

zungmann, 2016-12-22:
I tried the pid_autotune branch and it does not seem to work. Then I realized there is logical flaw in file heater.h, this lines:
/**
  When nothing is definded PID is active.
  Instead of #ifndef BANG_BANG on many places we use now #ifdef PID.
*/
#if ! defined(BANG_BANG) && ! defined(PID_AUTOTUNE)
  #define PID
#endif

When PID_AUTOTUNE is defined and BANG_BANG is not then PID never defined. SInce the heater.c is now using #ifdef PID then it practically just do BAN BANG.
I think we just can say:
#if !defined(BANG_BANG)
  #define PID
#endif

Wurstnase, 2016-12-22:
This part needs a complete rework. Since we changed the complete temperature readings, this won't work anymore when rebasing to current experimental.

But thanks for the hint. I keep that in mind when I start with a rework.

Wurstnase, 2017-01-04:
Reformating first post. Add the top 3 of my branches.