Traumflug/Teacup_Firmware #242: in configtool, saving a USB-checked configuration seems to fail

drf5n, 2016-10-06:
I was trying the current master branch configtool on a board.teensy++-v2.0.h file, I disabled the X axis minimum pin, and tried to save the board under a oard.Untitled.h name  and got this error:

```

$ pythonw configtool.py 
Traceback (most recent call last):
  File "/Users/drf/2014/RepRap/GIT/Teacup_Firmware/configtool/gui.py", line 240, in onSaveBoardConfigAs
    rc = self.pgBoard.onSaveConfigAs(evt)
  File "/Users/drf/2014/RepRap/GIT/Teacup_Firmware/configtool/boardpanel.py", line 239, in onSaveConfigAs
    rc = self.saveConfigFile(path)
  File "/Users/drf/2014/RepRap/GIT/Teacup_Firmware/configtool/boardpanel.py", line 265, in saveConfigFile
    v1 = pg.getValues()
  File "/Users/drf/2014/RepRap/GIT/Teacup_Firmware/configtool/communicationspage.py", line 59, in getValues
    result['BAUD'][1] = False
TypeError: 'tuple' object does not support item assignment

```

The file was not created and the tab on the header remains "board.teensy++-v2.0.h"

If I disable USB, the file will write, and the tab title will change.

Same error with board.teensy-v2.0.h

Traumflug, 2016-10-06:
Hah, you catched a bug. Nice! I can reproduce it by just loading and saving as-is the teensy++ board file.

drf5n, 2016-10-06:
I didn't think to try the simpler case, or maybe I did, but the lack of response while using the same name, was confusing .

Traumflug, 2016-10-06:
I think I've fixed it. My fault, I wrote that code. And a bug typical for untyped languages like Python.

It's on branch *issue-242*. Enjoy!

... and please report whether the bug is actually fixed.

drf5n, 2016-10-10:
I cherry picked it over to master and it compiles, uploads, runs, communicates, and moves the teensy++2.0 machine.   Thank you for the fix.

Traumflug, 2016-10-21:
Thanks for testing!

Moved this commit to *experimental*, then removed branch *issue-242*.

closed 2016-10-21 by Traumflug