Traumflug/Teacup_Firmware #244: Jogging and FRO

Dave3891, 2016-10-11:
Are there any branches that have jogging features and feed rate override? I would like to use this for a CNC router with a STM32 processor. I am currently using GRBL on a Arduino, but I want to switch to a ARM processor for more speed.

Traumflug, 2016-10-11:
Jogging features, like Nike shoes? Sorry, no shoes in the Teacup.

If you mean proper support of G0 and M220, then yes, of course. On branch *experimental*, IIRC, soon on *master*.

Dave3891, 2016-10-11:
I see the M220 speed override now, thank you.

In terms of jogging (I assume you know what I mean) something like the Galil motion cards that have a jog command to move the axis at a set speed until the jog stop command is issued which will decel the axis to a stop.

This can be done in GRBL, but it is kind of a hack. You issue a G0 with a large distance when a key is pressed and when the key is released you issue a stop command, wait for idle, then reset to clear the buffer.

phord, 2016-10-12:
Interesting idea. We do something similar internally with HOME commands.
What software issues the commands? Is it on the PC?

Phil

Dave3891, 2016-10-12:
Yes, it is on the PC with a custom GUI I wrote for a machine. I will take a look at the home command and see if I can modify it.

With the M220 override, this is not realtime correct? If I was cutting a line that was 10 inches and needed to change the speed would I need to wait for the next line for the M220 to be used?

phord, 2016-10-12:
I think M220 takes effect only on following commands.