Traumflug/Teacup_Firmware #245: temp.c

Wurstnase, 2016-10-19:
While thinking how to upgrade the temp to 12-14bit I see this code:
https://github.com/Traumflug/Teacup_Firmware/blob/experimental/temp.c#L246-L251

```
  if ((temp & 0x8002) == 0) {
    // Got "device id".
    if ((temp & 4) == 0) {
      temp = temp >> 3;
    }
  }
```

With this datasheet: https://cdn-shop.adafruit.com/datasheets/MAX6675.pdf

First `if`:
Bit15 is allway zero. A dummy sign. Why testing this?
Bit1 is low "to provide a device ID"?
Is it generally important to test this?

Second `if`:
We are testing Bit2. This will go high when the thermocouple opens. But should we also send a `TEMP_NOT_READY` in that case?

Traumflug, 2016-10-19:
Do you have such a sensor for testing? I don't. While I messed with some parts of this code while writing the SPI class, all the results are best guess.

If you see room for improvement, don't hesitate to change the code.

Wurstnase, 2016-10-19:
Unfortunately not. However, pushed some new code.
https://github.com/Traumflug/Teacup_Firmware/commit/ec969408b48f1