Traumflug/Teacup_Firmware #246: Misplaced paren in configtool/thermistortablefile.py

drf5n, 2016-10-24:
https://github.com/Traumflug/Teacup_Firmware/blob/master/configtool/thermistortablefile.py#L169

If you attempt a Steinhart-Hart table in the configtool with parameters (4700, 25, 100000, 209, 475, 256, 201) it fails with a:

    ...
     File "/Users/drf/2014/RepRap/GIT/Teacup_Firmware/configtool/thermistortablefile.py", line 169, in SteinhartHartTable
        (i, int(t * 4), int(delta * 4 * 256), c, int(t), int(round(r))),
      TypeError: not enough arguments for format string

If you move the last paren from line 169 to line 170, the 'delta' parameter is included in the interpolation list.

Traumflug, 2016-10-24:
Good catch, confirmed.

Made a commit according to your description and put it on branch *issue-246*, it fixes the issue here. Can you confirm this to be the right solution?

drf5n, 2016-10-26:
I cherry-picked the commit over to my setup and can confirm that it fixes the error and builds for me.

Traumflug, 2016-10-26:
Thank you very much for the verification. Picked this commit to *experimental* and removed *issue-246*.

closed 2016-10-26 by Traumflug