Traumflug/Teacup_Firmware #247: M220 should not apply while searching endstops (G28)

Wurstnase, 2016-11-08:
This should fix this:
https://github.com/Traumflug/Teacup_Firmware/tree/multiplier_endstop_fix

Traumflug, 2016-11-08:
Looks good. My personal nitpicking applied, picked to *experimental*, branch removed.

Thank you very much!

Wurstnase, 2016-11-08:
Your welcome. It's sometimes a good idea to test a lot of things _not_ in a real machine.
I found this issue while playing with M220 S1000. I guess without that huge value you could not notice it.

phord, 2016-11-08:
I don't usually home during a print, and I don't usually adjust M220 when I'm not printing.  I'm sure I might never have run into this.  I'm glad you caught it. :+1: 

Did you find it on a real machine?  Did it break anything? :fearful:

Wurstnase, 2016-11-09:
I was playing with pure steppers on my desk. Nothing breaks. Only my stepper can't follow the step impulse.

closed 2017-07-18 by Wurstnase