Traumflug/Teacup_Firmware #248: dda lookahead overflow

Wurstnase, 2016-11-13:
Hi,

I'm searching for something in dda lookahead which will overflow. This part should be dependent on steps per mm and the feedrate.

My steps per mm is 1280. At 3037 mm/min I can travel without any issue. At 3038 mm/min something will overflow. But I don't have any clue where or what. There should be some rounding in it. 
1280 * 3038 / 60 ~ 2^16.

I replaced already muldiv and muldivQR with 64bit instructions. But no health.

Wurstnase, 2016-11-13:
Very strange thing. I don't have any idea what I've changed, but it's gone for now. When I will find this issue again, maybe I know why this happened and I will write into this issue. Until that, closed for now.

Wurstnase, 2016-11-14:
It's maybe not the lookahead.

I activate debug DDA and when using some 'special' values for the feedrate (dependent on what code i've using, need to investigate more for this), the DDA message breaks before reaching lookahead.

One issue could be overflowing the rampup_steps alogrithm with acc_ramp_len().
As i mentioned here before: https://github.com/Traumflug/Teacup_Firmware/commit/22120151a78df0f
With acceleration > 596, this will overflow:
```
((uint32_t)7200000UL * ACCELERATION) / steps_per_m

2^32 / 7200000UL ~ 596
```

Wurstnase, 2016-11-14:
Disassemble of ACC 3000
-> 7200000UL * 3000 = 125163520

Wurstnase, 2016-11-14:
Yes, this was the problem. My dry print works now without issues.
I take only this commit: https://github.com/Traumflug/Teacup_Firmware/commit/15248ba98c

Wurstnase, 2016-11-14:
```
start
Print started at: 19:26:41
Setting print speed factor to 250%.
stop
Print ended at: 21:45:11 and took 2:18:29
```
Yeah :D

Traumflug, 2016-11-14:
Let's cross fingers this code works as well as it looks!

All three commits on *acc_ramp_len* picked to *experimental*, then branch removed.

Wurstnase, 2016-11-14:
Oo :+1: 

Two things.
https://github.com/Traumflug/Teacup_Firmware/commit/5b11a391551 :
For total independence of X and Y it needs some more code. Afaik also for lookahead. But it should be a first commit in that way.

https://github.com/Traumflug/Teacup_Firmware/commit/cd66feb8d1 :
Here it is important to `x = x / something * somethingelse` because `x /= something * somethingelse` will overflow to early.

Wurstnase, 2016-11-15:
https://github.com/Traumflug/Teacup_Firmware/blob/master/dda_lookahead.c#L216
This line should be:
```
this_F = muldiv(current->fast_um, this_F, current->distance);
```

Wurstnase, 2016-11-15:
Wrong name for the branch, anyhow this was an old issue no one mention before.
https://github.com/Traumflug/Teacup_Firmware/tree/issue-69b

Wurstnase, 2016-11-15:
Not very sure. But do we need atomic copies of the 'current' values? They all are just created by dda_create() and will also only modified in dda_create() and this will call dda_join_moves().

'prev' can change, but not 'current' or do I miss something?

Traumflug, 2016-11-15:
I assume you talk about dda->c_min. If these values don't change during a move, they don't need to be atomic. They can change when switching from one move to the next. But then there are either acceleration ramps in between or the values are similar enough to allow ignoring the difference.

That leads to the conclusion that there's no need to compare move_c with c_min at all. Either the state is in an acceleration/deceleration ramp, where it's calculated, or move_c can be set to c_min without tests.

Wurstnase, 2016-11-16:
I agree with that.
dda->n needs to be atomic, but not dda->c. Also changes of c are very small and should have never a big impact.

Now looking into dda_lookahead.c in dda_join_moves().

At start we catch a lot of values atomically. But in the end we only write new values, when the ID is the same as in the beginning. So we could read all values non atomically in the beginning?
```
...START with dda_join_moves...
  if (prev->live == 0) {
    // Perform an atomic copy to preserve volatile parameters during the calculations
    ATOMIC_START
      prev_id = prev->id;
      prev_F = prev->endpoint.F;
      prev_F_start_in_steps = prev->start_steps;
      prev_F_end_in_steps = prev->end_steps;
      prev_rampup = prev->rampup_steps;
      prev_rampdown = prev->rampdown_steps;
      prev_total_steps = prev->total_steps;
      crossF = current->crossF;
      this_id = current->id;
      this_F = current->endpoint.F;
      this_total_steps = current->total_steps;
    ATOMIC_END
...
... do some stuff
...
if(prev->live == 0 && prev->id == prev_id && current->live == 0 && current->id == this_id)
```

Wurstnase, 2016-11-16:
In line 233 there is a break condition. This should need than an extra compare with the ID. But all-in-all we could remove a lot of atomic ops.

Wurstnase, 2016-11-16:
Here is an high experimental version of my idea: https://github.com/Traumflug/Teacup_Firmware/commit/ef8aad3a9838

Any idea how to provoke the dda to be too slow for lookahead?

Traumflug, 2016-11-16:
> Any idea how to provoke the dda to be too slow for lookahead?

Insert delays. Like delay_us() or delay_ms(). They delay by literally wasting computing power.

phord, 2016-11-16:
Atomic operations aren't bad if they are kept relatively brief.  You do need to protect the assignment or comparison (generally) of any 16-bit or larger value. 

Your experimental version tries to do too many things in one commit.  As you noticed, it seems you need to protect only the DDA->id values atomically.  The rest can be used non-atomically[1] and do not need to be copied inside the ATOMIC section.  Here's my attempted simplification of your experiment: 0d26b02

My change (mysteriously?) adds 8 bytes to the program size, but you can counter this with your local-variables-reduction in a separate change.

I don't think it is safe to do away with the checks on current_id.  Maybe you can convince me otherwise.

[1]  You should examine the function carefully to make sure it does not segfault or otherwise do nasty things if it gets garbage from the now non-atomically used data.  On a microcontroller we generally don't need to fear DIV_BY_ZERO or other math faults caused by garbage-in, but other/future platforms might be less tolerant.

Wurstnase, 2016-11-16:
You are right. Doing less in one commit would be better.

[1] Thanks for that hint. I'v read now something about it. We have only 3 'divisions'. The `muldiv`. But the divisors were never atomic in the past.

current_id don't need to be atomic. We are creating just in this moment the current. This will never start before` dda_create()` has finished and `dda_start()` is executed afterwards. In theory we could inline `dda_join_moves()`.

We only need also `prev->total_steps` and `prev->start_steps` because they are used in the break condition at line [233](https://github.com/Traumflug/Teacup_Firmware/blob/experimental/dda_lookahead.c#L233).

I made now a first test. I've insert a 100ms delay after the atomics. I've tested the break-condition, current->id and prev->id. As I've thought, current->id never changed. break-condition most time at infill, where you have 180° direction changes and previous will be changed at near any time in a circle.

Wurstnase, 2016-11-16:
```
b:272 c:0 p:3243
b:272 c:0 p:3243
b:294 c:0 p:3243
change from infill to high segmentation circle
b:295 c:0 p:3318
b:295 c:0 p:3401
```
b++ if break condition
c++ if current->id change || current->live
p++ if prev->id change || prev->live

phord, 2016-11-16:
Ah, you're right.  Currently dda_join_moves is only called from dda_create and `current` is always an unstarted movement.  But the comments on both locations (caller and callee) suggest this is expected to change in the future when dda_join_moves learns to comb through the queue improving the speeds of earlier movements as well.

I'm not sure about the need for the atomic checks at the break condition because A) it should still be garbage-tolerant, and B) it could be enhanced to return also if prev->id has changed if you are concerned about its use as a performance optimization.  But maybe it's better to be safe if there is any question.

Traumflug, 2016-11-16:
> As you noticed, it seems you need to protect only the DDA->id values atomically.

dda->id is an 8-bit value, so no need at all to be atomic during reads.

> We have only 3 'divisions'. The muldiv. But the divisors were never atomic in the past.

With atomic reads one can trust to read sane variables. Without atomic reads, these variables can contain garbage, in the rare case that dda_start() switches to the next movement right during these reads.

That said, if processed variables are allowed to contain garbage, there's no need to copy them at all for reading. Only variables written later have to be kept locally.

Wurstnase, 2016-11-16:
Garbage values are not really a problem. We first saves the ID and check in the end, that the IDs should match. Only then we are writing the values atomically.

At the break condition we could also check the IDs, if we will later have more rounds of dda_join_moves(). In that case we don't need to save the two `prev->xx_steps` values.

Wurstnase, 2016-11-17:
I went yesterday a more radical way and removed any atomic in the beginning. As @Traumflug mentioned, the ID is 8bit and anyway atomic.

So I only test the IDs just before the break condition and in the end before writing atomics.

I will test this in the afternoon also with an Arduino Mega with some huge delays.

phord, 2016-11-17:
You do need the atomic at the end.

Wurstnase, 2016-11-17:
> You do need the atomic at the end.

Yes, sure. I mean the complete atomic block in the end. This will stay atomically.

Wurstnase, 2016-11-17:
Another point why not testing current->id:
All values are garbage when prev->id is not valid. And prev->id is always before current->id.

Wurstnase, 2016-11-17:
Some cleaning work and more comments of my ideas and why I'm doing it can be found here:
https://github.com/Traumflug/Teacup_Firmware/tree/less_atomics3

Wurstnase, 2016-11-18:
gcc 6.1.1 and gcc 4.9.2 will produce a lot differences of program size.

```
               gcc 4.9.2        gcc 6.1.1
experimental:   23486            23866
less_atomics3:  23478            23846
--------------------------------------
diff:              -8              -20
```