Traumflug/Teacup_Firmware #249: BANG_BANG Not defined

youssefaly97, 2016-11-15:
In the master branch, BANG_BANG, BANG_BANG_ON, BANG_BANG_OFF are not defined. As a temporary fix I just added these to heater.h 

#ifndef BANG_BANG
#define BANG_BANG
#endif

#define BANG_BANG_ON -2
#define BANG_BANG_OFF 2

I was only checking the binary size for a specific configuration, so the values I entered for bang bang on and off are probably wrong, considering I've always preferred PID.

Traumflug, 2016-11-15:
These values should be part of your printer configuration file, config/printer.xxx.h. I just checked, they're there. Also, Configtool writes them.

youssefaly97, 2016-11-16:
Well I checked, BANG_BANG is there, the On and Off are commented out. Still thought why does it not recognize it ?

Wurstnase, 2016-11-16:
Maybe it would help when activating BANG_BANG in configtool that there are some default values.

```
heater.c: In function 'heater_tick':
heater.c:186:17: error: 'BANG_BANG_OFF' undeclared (first use in this function)
    pid_output = BANG_BANG_OFF;
                 ^
heater.c:188:17: error: 'BANG_BANG_ON' undeclared (first use in this function)
    pid_output = BANG_BANG_ON;
```

youssefaly97, 2016-11-16:
Well, I haven't given it much thought. I just thought it's a small bug, report it.