Traumflug/Teacup_Firmware #250: Hello Matt Gilbert!

Traumflug, 2016-11-18:
Matt modified a printer for creating PCBs using the 'scratch-n-etch' method and used a modified Teacup as firmware. Let's add him as contributor so he can bring his enhancements back in here.

@mattgilbertnet, the rules are simple here: don't touch *master* or *experimental*; instead create as many topic branches as you feel need for. Once these topic branches find agreement, they get cherry-picked or fast-forward-merged onto *experimental*. On how to rebase for following recent enhancements, see e.g. issue #81 or general documentation about Git topic branches.

Traumflug, 2016-11-18:
Ah, forgot the RepRap wiki forum thread: http://forums.reprap.org/read.php?131,723945,724060 :-)

mattgilbertnet, 2016-11-18:
Thanks Traumflug! I'm pretty sure the changes will be very minimal, and I'm excited to contribute!