Traumflug/Teacup_Firmware #252: Can't compile Teacup Firmware for LPC1114

MrSliff, 2016-12-03:
Hello,

since 3 days im trying to compile the Firmware for the Gen7-ARM Board. But I have the same issue on 3 different Systems (Ubuntu Mate on BananaPi, Debian on VM, Ubuntu on VM). 

Every make command on those Machines gives me this Error:

```
rootbox@BananaPi:~/Desktop/Teacup_Firmware-master$ make
  CC        build/mendel.o
/home/rootbox/Desktop/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: 1: /home/rootbox/Desktop/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: Syntax error: word unexpected (expecting ")")
Makefile-common:88: recipe for target 'build/mendel.o' failed
make: *** [build/mendel.o] Error 2
```

I've tried several ways of putting folders somewhere, change access policies, ande some else. No luck since now. 

Its a bit frustrating.

Also Configtool gives me errors on Mate and Debian when trying to save the config.h. Is it right that there are just the includes of the board.h and printer.h inside the config.h? Maybe thats the Problem?

Another Problem on Debian: There's an error when starting configtool.py, but it starts. But when opening the calculators for the steppers, there's no close button for them. Not a problem for me, since it works on Ubuntu.

Can someone PLEASE help me to get this working? It would be better to have the ability to compile it by myself due to testing adjustments and making addons for SPI things.

Traumflug, 2016-12-03:
> Also Configtool gives me errors on Mate and Debian when trying to save the config.h.
> [...]
> There's an error when starting configtool.py

The only way we can help here is if you tell us these errors. Like copy&pasting them into here, like making a screenshot or a photograph and inserting that here (see "Attach files..." at the bottom of the edit box here).

Even if stuff starts working at some point, reporting such errors is still a good idea, because if we don't fix this, the next user will run into this as well.

> I've tried several ways of putting folders somewhere, change access policies, ande some else. No luck since now.

It's a syntax error. Which means, the *content* of one of the files doesn't meet what the compiler expects. No surprise things don't change if you move them around, because such maneuvers don't change file content.

Here's a config.h like it should look like:
```
// Configuration for controller board.
#include "config/board.gen7-arm.h"

// Configuration for printer board.
#include "config/printer.mendel.h"
```

MrSliff, 2016-12-03:
Ok, so my config.h looks like this.

The Error in Configtool GUI comes up, when i want to save config.h. Last save attempts didnt threw an error. But it was something like "Teacup can't overwrite files brovided by itself". Not exactly the error message, but something similar with same meanings.

On debian there was an error of a missing package, i think i fixed it with installing it. Sorry, i don't know anymore which package it was, because there no error anymore. Just the missing "Close" button in Inside Application windows. I can check with a fresh install later.

BTW: Ubuntu and Debian are 64bit OS except the BananaPi.

But there remains the compiler error. Where should i start to search?

Traumflug, 2016-12-03:
> But it was something like "Teacup can't overwrite files brovided by itself".

Thanks. Saving the board/printer with a different name makes this error going away. The point is, if Configtool allows to overwrite distributed files, these distributed files get lost, so a user can't start over.

What would be your intuitive expectation? Renaming the configuration right when loading it? Renaming it only when the user changed something? Allowing to overwrite distributed files, because a user typically starts over by throwing all Teacup files away first?

Traumflug, 2016-12-03:
> But there remains the compiler error.

I'm a bit lost, too, because unchanged files compile here just like they should. Also, the given error is in line 1, which contains no code.

Hmm. Could you tar up exactly what you have and send it to me by email? Tarring up goes like this (from the parent folder):

    tar -cvzf Teacup_Firmware.tar.gz Teacup_Firmware/

MrSliff, 2016-12-03:
Ok i've made a coopy, you can Download it here:

https://www.dropbox.com/s/728a0b6vhpxm6q8/teacup.tar.gz?dl=0

Im gonna try to compile with standard settings too.

Traumflug, 2016-12-03:
Thanks. After changing the TOOLCHAIN variable in the Makefile it builds mendel.o just fine.

    TOOLCHAIN = arm-none-eabi-

This is a hint that the compiler might be at fault. Can you build anything else?

After that, building dda.o fails. I see that Y_STEP pins are commented out. Did you perhaps try to save the board file? This doesn't work for ARM based controllers, yet. You can save config.h and a printer configuration, but not the board file.

MrSliff, 2016-12-03:
Finally, i've got it to work!

I've just downloaded the non-install tar for arm-embedded and tried with this before.

Now i've installed the toolchain with this guide from Teacup-Wiki Site:
`sudo apt-get purge gcc-arm-none-eabi # ignore errors
sudo apt-get purge binutils-arm-none-eabi # ignore errors
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
sudo apt-get install build-essential gcc-arm-embedded lpc21isp`

And just copied the Makefile-ARM to Makefile and made no changes to the file.

Then it worked with standard settings. Now i just have to configure the Thermistors and the Stepper Motors and i can go to a first function test.

Did this with a fresh Ubuntu 16.10 64bit install on a Virtual Machine.

No more errors when starting configtool.py and compiling software!

Thank you for your Support Traumflug!

Traumflug, 2016-12-03:
> Now i've installed the toolchain with this guide from Teacup-Wiki Site [...] Then it worked with standard settings.

The million dollar question is: how do we get people into following these instructions first, before experimenting with whatever funny ideas they have in mind?

MrSliff, 2016-12-03:
Maybe a bit more structured and a bit better Documentated Howto. This is not meant to be a harsh critic, I appreciate your work!  

There maybe should be an extra section for the ARM-based board in the wiki, because it's just well explained for AVR and with or without Arduino IDE. Or something like an IDE-Project with which you can compile the Firmware. For example Coocox or Eclipse IDE. Maybe that would be a bit easier to handle. 

Because I ran into problems, ive decided to search the internet.

One more question: 
Since I cant change settings in the board section, can I change settings in the "thermistortable.h" and are those settings included when I build the firmware?

Traumflug, 2016-12-03:
> Since I cant change settings in the board section, can I change settings in the "thermistortable.h"

Yes. Change it in Configtool without saving the board file. thermistortable.h is written together with config.h, so save that.

You can edit the board file manually, with a text editor. Configtool just can't save it, because pin names are (still) hardcoded for AVR inside Configtool and ARM pins have different names.

MrSliff, 2016-12-03:
Thank you very much!

If I have time, i'll take a deeper look into it, but at the moment i'm overloaded with work. Sorry for that. But maybe i can help a bit with modifying. I have some knowledge, but as I said, I dont have a academic grade in programming, just knowledge learned by myself and at work.

MrSliff, 2016-12-03:
There are some thing to do for me related to stepper drivers in combination to the PSU. I don't know if we should talk about it here or somewhere else.

It's about separating PS_ON and Stepper-Driver Enable pins. I've already done that, because PS_ON is Low-Active and my Stepper Drivers are High-Active. Was a bit confusing when I made the first test with Canned Cycle... The PSU turned on, but the Steppers were not moving :)

So I need help with configuring the STEPPER_ENABLE Pin that it puts out a high level. I already checked and tested some settings in pinio.h but the pin doesnt put a high level when the drivers should be enabled. I figured out, that in standard configuration, the voltage level for stepper enable depends on the PS_ON pin since it has a high level when not pulled to ground.

Could you give me a hint how to configure this?

```
  /// Common Stepper Enable.
  #ifdef STEPPER_ENABLE_PIN
    SET_OUTPUT(STEPPER_ENABLE_PIN);
    #ifdef STEPPER_INVERT_ENABLE
      WRITE(STEPPER_ENABLE_PIN, 0);
    #else
      WRITE(STEPPER_ENABLE_PIN, 1);
    #endif
  #endif
```

For now i just pulled up the enable pins to 3.3V with a 1k resistor and theyre only enabled when the PSU is turned on. Thats a workaround i can live with, but in my opinion it would be better to have as less workarounds as possible on the board. I'm lucky that there are some unused Pins left on the chip, so I can do a workaround for the PS_ON pin.

triffid, 2016-12-04:
Looks like you just need to define STEPPER_INVERT_ENABLE in your config
somewhere

MrSliff, 2016-12-04:
I already did that, but it loooks like it doesnt work.

MrSliff, 2016-12-04:
Here's my actual firmware config. 

[https://www.dropbox.com/sh/20p1tavjqok4dcf/AAAEWApe7CKWEt8hhmsTm0Swa?dl=0](url)

PS_ON_PIN and STEPPER_INVERT_ENABLE are deactivated. As I said, I've tested that case, but the Pin never pulls to high level, even with or without STEPPER_INVERT_ENABLE.

Wurstnase, 2016-12-04:
Gen7 ARM can't handle active high for stepper driver. The stepper enable is connected to the PS_ON. So maybe someone has an idea to separate them?!?

MrSliff, 2016-12-04:
Okay. 

I habe another idea for that: I could cut THW Trave between the controller and the enable Pins, leave the PS_On connected to the controller and THW pulled up enable Pins. That should work for my solution

MrSliff, 2016-12-04:
Oops sorry, autocorrection.

Traumflug, 2016-12-04:
I've looked the pin up in the data sheet. It's PIO0_5, which is one of the two pins also serving I2C. Which means, they're open drain and have no internal pullup resistor. See chapter 7.4.11 in the LPC11xx user manual, page 78, and later in the I2C chapter.

Simply applying an external pullup resistor should work. 10K, 20K, 100K, something like this.

MrSliff, 2016-12-04:
We're talking about the `#define STEPPER_ENABLE_PIN       PIO0_4`?

I think PIO0_4 is connected to PS_ON and the EN-pins on the stepper drivers. Same in the schematic for the Gen7-ARM board.

closed 2016-12-05 by MrSliff

Traumflug, 2016-12-05:
Does this closing mean "it works now" or does it mean "I've given up"? And a "thank you" is way too much expectation, of course.

MrSliff, 2016-12-05:
Damn sorry! I can remember that I clised this today morning with a thank you... Maybe i was still dreaming :)

Im sorry for that!

So it works now with the modifications above. I can give you a small documentation with pictures if you want.

Anyways, thank you again for your support! I can't believe that i didnt writeor post that this morning... Sorry.

Traumflug, 2016-12-05:
Glad to read it was closed for "it works now".

If you want to write a few sentences about how to get drivers with Enable High working, please go ahead: https://www.reprap-diy.com/gen7_and_enable_high_stepper_drivers?do=edit

MrSliff, 2016-12-05:
Sure, gladly! I'll check when I have free time, maybe wednesday.