Traumflug/Teacup_Firmware #255: Get rid of dda->delta_um

Traumflug, 2016-12-10:
Storing these four 32-bit values for almost nothing always bugged me, so let's take the plunge and remove them entirely.

These values were introduced together with lookahead. However, lookahead code eventually switched from micrometer based to step based calculations, so these values became almost obsolete. But not entirely.

Code simply doing the necessary things is on branch *delta_um*. Regression tests pass. Questions to answer:

- How much does the calculation of dda->crossF deviate from the previous algorithm?
- Does it work in all corner cases (zero steps, axis steps == total steps, etc.)

Wurstnase, 2016-12-11:
currF and prevF are loosing its sign. 
delta velocity calculations could be wrong.
Before 
currF = -100
prevF = 90
currF < prevF
Now
currF = 100
prevF = 90
currF > prevF

Wurstnase, 2016-12-11:
```
G21
G90
G1 X4 F400
G1 X0 F390
G1 X4 F390
G1 X0 F400
M2
```
experimental:
[image]

delta_um:
[image]

Wurstnase, 2016-12-11:
delta_um2:
[image]

Wurstnase, 2016-12-12:
The direction flag should be moved with this example out of the all_flags section in dda.h. This is only 32bit.

Wurstnase, 2016-12-12:
DDA: get rid of dda->delta_um[].
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22138 bytes      155%       73%       35%       18%
   Data:   1256 bytes      123%       62%       31%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```
DDA: new direction and use it for dv in dda_lookahead
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22208 bytes      155%       73%       35%       18%
   Data:   1280 bytes      126%       63%       32%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```
DDA: move direction out of allflags union. (struct DDA)
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22212 bytes      155%       73%       35%       18%
   Data:   1280 bytes      126%       63%       32%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```
DDA: We don't need to reset the directions in dda_queue.
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22196 bytes      155%       73%       35%       18%
   Data:   1280 bytes      126%       63%       32%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```
DDA: use axis_uint8_t also for debounce_count
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22196 bytes      155%       73%       35%       18%
   Data:   1281 bytes      126%       63%       32%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```
dda.c: moving code to reduce size.
```
ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  22182 bytes      155%       73%       35%       18%
   Data:   1281 bytes      126%       63%       32%       16%
 EEPROM:     32 bytes        4%        2%        2%        1%
```

Traumflug, 2016-12-12:
Thanks for analysis code. Looks like I missed something with this quick shot :-/

While binary size increase for switching from single bit flags to an array of bytes is bearable, it appears to cost 12 clock cycles performance in dda_start() (see maximum times, compare to https://github.com/Traumflug/Teacup_Firmware/commit/aa0ef9a3e00fe81e99bc6e8e9bce642d73681808) :

```
Commit 'DDA: We don't need to reset the directions in dda_queue.'

ATmega sizes               '168   '328(P)   '644(P)     '1280
Program:  17948 bytes      126%       59%       29%       14%
   Data:   1968 bytes      193%       97%       49%       25%
 EEPROM:     32 bytes        4%        2%        2%        1%

short-moves.gcode statistics:
LED on occurences: 888.
LED on time minimum: 202 clock cycles.
LED on time maximum: 392 clock cycles.
LED on time average: 232.336 clock cycles.

smooth-curves.gcode statistics:
LED on occurences: 23617.
LED on time minimum: 220 clock cycles.
LED on time maximum: 435 clock cycles.
LED on time average: 255.233 clock cycles.

triangle-odd.gcode statistics:
LED on occurences: 1636.
LED on time minimum: 220 clock cycles.
LED on time maximum: 392 clock cycles.
LED on time average: 245.59 clock cycles.
```

I picked up some of your ideas and made another commit on top of *delta_um*, keeping performance. Better now?

Wurstnase, 2016-12-12:
> cost 12 clock cycles performance

This is a pity. I don't like those axis-dependent naming. 
Anyhow, the code looks good. I will test this later, but I don't see any changes for now.

You could pick https://github.com/Traumflug/Teacup_Firmware/commit/c2e38e711c51792077. It is better to use the variables testing in the `if`-condition just behind it. So the controller don't need to push and pop them later.

Wurstnase, 2016-12-12:
[image]

Traumflug, 2016-12-15:
Looks good, right? Squashed both commits and forwarded it to *experimental*.

Wurstnase, 2017-07-18:
In the end we had revered some commits. Everything works properly. 

Just hang around mikrocontroller.net and build the avr-gcc 8.0.0 and test the latest experimental.
Codesize can shrink up to 500 bytes. Pretty impressive.

Speed can also increase up to 11 cycles.

For example:smooth-curves.gcode statistics:
**avr-gcc (GCC) 4.9.2**
```
smooth-curves.gcode statistics:
LED on occurences: 22589.
LED on time minimum: 209 clock cycles.
LED on time maximum: 521 clock cycles.
LED on time average: 276.729 clock cycles
```

**avr-gcc (GCC) 8.0.0 20170718 (experimental)**
```
smooth-curves.gcode statistics:
LED on occurences: 22589.
LED on time minimum: 209 clock cycles.
LED on time maximum: 510 clock cycles.
LED on time average: 277.884 clock cycles.
```

Anyhow. I think we can close this one here soon.

triffid, 2017-07-19:
Heh so the peak drops by 11 cycles but the average goes up by one cycle?
Curious, I guess it's made a different branch optimization somewhere

Wurstnase, 2017-07-19:
They (avr-gcc guys) doing some stuff for the interrupts. Pre-8 there will be some unnecessary push/pops. 
I can't find any different in the time-ISR between 4.9.2 and 8.0.0. Here is a small example from mikrocontroller.net:
```
uint8_t rxd;
uint8_t txd;

ISR(SPI_STC_vect)
{
  rxd = SPDR;
  SPDR = txd;
}
```

```
push    r1
push    r0
in      r0, 0x3f  ; 63
push    r0
eor     r1, r1
push    r24
in      r24, 0x2e  ; 46
sts     0x0101, r24
lds     r24, 0x0100
out     0x2e, r24  ; 46
pop     r24
pop     r0
out     0x3f, r0  ; 63
pop     r0
pop     r1
reti
```

vs.

```
push    r24
in      r24, 0x0f
sts     0x0063, r24
lds     r24, 0x0062
out     0x0f, r24
pop     r24
reti
```