Traumflug/Teacup_Firmware #256: Hello Friend. Help for teacup+Schild CNC V3

Ve-nik, 2016-12-15:
The situation is this: there Schild CNC V3, there are two backup Spend A4 and A5, I want to tie them thermistor table and a heater in the MOSFET, but as I can not find where to do this firmware. In config.h I have added:
/ ** \ Def TEMP_SENSOR_PIN
Temperature sensor pins a user should be able to choose from in configtool.
All commented out.
* /
// # Define TEMP_SENSOR_PIN AIO1
// # Define TEMP_SENSOR_PIN AIO2
// # Define TEMP_SENSOR_PIN AIO3
// # Define TEMP_SENSOR_PIN AIO4
// # Define TEMP_SENSOR_PIN AIO5
and has added:
/ ** \ Def HEATER_PIN
Heater pins a user should be able to choose from in configtool. All
commented out.
* /
// # Define HEATER_PIN AIO2
// # Define HEATER_PIN AIO4
// # Define HEATER_PIN AIO5
Launch ConfigTool, they appeared, but did not work. I climb through the Arduino section gcode_process.s and see there:

sase 140:
//? --- M140: Set heated bed temperature ---
//? Undocumented.
#ifdef HEATER_BED
if (! next_target.seen_S)
break;
temp_set (HEATER_BED, next_target.S);
#endif
break;
What am I doing wrong, please help. Or I did not go the right way?

Wurstnase, 2016-12-15:
Can you upload your complete new board.xxx.h file?

Ve-nik, 2016-12-15:
I apologize if  do something wrong.
[Teacup_Firmware Venik.zip](https://github.com/Traumflug/Teacup_Firmware/files/654498/Teacup_Firmware.Venik.zip)

Wurstnase, 2016-12-15:
Looks like you are doing it right. The pins you take for the heater don't have PWM but this shouldn't change anything for a test.

So what happened? Or what not?

Ve-nik, 2016-12-15:
The firmware is compiled, but the program does not work.

Wurstnase, 2016-12-15:
Does not work can be everything. You should describe exactly what happened. Happened nothing? No serial connection? No feedback at all? Or does the heater do nothing?

Also, maybe a picture of your setup could help. Circuit of your heater-setup and thermistor-setup.

Ve-nik, 2016-12-15:
Thermistor  hotend work. Thermistor and the hot bed, no. Tried to change only the hot bed for free port AIO0, too, failed. The assembled circuit, a copy of the RAMPS, electronics 100% working.
[image]
[image]

Wurstnase, 2016-12-15:
Are you sure, that the pins are connected to the Uno? Can you measure the resistance between the pin on the CNC Shield?

Ve-nik, 2016-12-15:
My assumption that the problem is with the port assignment on the heated hotbed. UNO and CNC V3 resistance of 0.01 ohms.

Ve-nik, 2016-12-15:
Maybe can I somehow remap the RX and Tx, but how?

Wurstnase, 2016-12-15:
> UNO and CNC V3 resistance of 0.01 ohms.

That is ok.
Can you please configure the heated bed to a fan? And check if you can set this fan with M106 S255 on?

> Maybe can I somehow remap the RX and Tx, but how?

No you can't remap the serial connection.

Ve-nik, 2016-12-15:
Oh yes, it's an idea! Today, I will try. Thank you.

Ve-nik, 2016-12-16:
Thank you, dear friend. Your advice helped in the decision. The problem was incorrectly marked contacts on the shield (SCL and SDA).I'm in the firmware all pins made fan, and on the next run commands M106 S255  I found the truth.

Wurstnase, 2016-12-16:
Nice. Which pins are correct now?

Ve-nik, 2016-12-16:
SLC-AIO5 and SDA-AIO4  inverted.

Wurstnase, 2016-12-16:
Thanks for the Info. I have also the CNC Shield V3. But currently I don't use the SCL and SDA pins.

If you want 'more' powerfull controller, you could replace the Arduino Uno with a Nucleo F411RE. I'm working on a port since one year.

Ve-nik, 2016-12-16:
NUCLEO-F411RE excellent platform 32bit perhaps necessary preobresti and study quietly. 8-16 bits already obsolete.

Wurstnase, 2017-02-27:
I close this, because it's solved.

closed 2017-02-27 by Wurstnase