Traumflug/Teacup_Firmware #257: dda.c: move duration

Wurstnase, 2016-12-19:
New issue for new things instead of messing around this issues: https://github.com/Traumflug/Teacup_Firmware/issues/233#issuecomment-267002757
While working on acceleration temporal I made also a rework of this TODO.

https://github.com/Traumflug/Teacup_Firmware/blob/experimental/dda.c#L316
```
    // TODO: instead of calculating c_min directly, it's probably more simple
    //       to calculate (maximum) move_duration for each axis, like done for
    //       ACCELERATION_TEMPORAL above. This should make re-calculating the
    //       allowed F easier.
```

move_duration can be calculated in ticks and with that we can calculate dda->c directly. I test it with smooth-curves.gcode. There we call 42 times the dda_create(). Minimum calculation times increased by 200 ticks, but the average will decrease by 200 ticks with the newer version. (https://github.com/Traumflug/Teacup_Firmware/commit/856a07d33644e68d7)

Wurstnase, 2016-12-19:
Next step is to change the calculation of c_min. We know the max feedrate. We know the steps/m. So we also know c_min for each axis. Precalculate it and use it as c_min. Then calculate the new endpoint.F.

For a distance < 2^16 we can then simplify it to:
```
if (delta[i] * endpoint.F > distance * maximum_feedrate)
 md_axis = i;

c_min = c_min_P[md_axis];
endpoint.F = (F_CPU * 1000 * 60) / (c_min * steps_per_m[dda->fast_axis]);
```

If the distance is > 2^16 we could loop through with `>>1` until it's below.

Wurstnase, 2016-12-19:
So much effort... For... nothing...

Experimental is 168 bytes smaller:
```
LED on time minimum: 5495.68 clock cycles.
LED on time maximum: 5520.48 clock cycles.
LED on time average: 5509.59 clock cycles.
```
Last move_duration:
```
LED on time minimum: 5470.88 clock cycles.
LED on time maximum: 5586.2 clock cycles.
LED on time average: 5492.9 clock cycles.
```

Traumflug, 2016-12-19:
Not nothing. You've learned something.

Wurstnase, 2016-12-19:
Sure ;) muldiv is very expensive.

Also:
```
git checkout something~n
git rebase HEAD something
```
is sometime simpler than interactive rebase.