Traumflug/Teacup_Firmware #258: Extruder motor won't move properly

MrSliff, 2016-12-21:
Hey, its me again :)

First I want to say sory, that I didnt have time yet to describe the Pullup modification for HIGH_Level enabled stepper drivers. im going to do it until the end of the year. Promise!

So, while testing I've ran into a problem with my extruder motor. It looks like doesnt move properly. It wants to, put it doesnt.

So I did some testing. My first thoughts were electrical problems, so I switched the driver to the Z-Axis. Same Motor was moving, so the driver and the Motor are not defective.

Then I tried some current and decay adjustment on the driver. That didn't help too.  Made some short circuit measurements. Everything fine too.

So I decided to switch the step pin to another driver. After that, the problem occured on the other driver.  

It looks like the step generation for the extruder driver is different than for the axis drivers, isnt'it? After watching in the source file where step signal generation is done, it looks like it's done the same way for every stepper driver (axis and extruder).

Could you give me an advise where I can find the right function/interrupt for the extruder step generation? Or maybe how to change it?

Here you can find the recent config of my firmware:

https://www.dropbox.com/sh/20p1tavjqok4dcf/AAAEWApe7CKWEt8hhmsTm0Swa?dl=0

Geetings

MrSliff

Wurstnase, 2016-12-21:
Teacup configs the stepper with steps per meter. Not steps per millimeter!
`#define STEPS_PER_M_E            291` should be `#define STEPS_PER_M_E            291000`

MrSliff, 2016-12-21:
Damn!

Thank you for that. I can remember that i calculated it, but I didnt think about calculating it for metres. Even if I knew that.

MrSliff, 2016-12-21:
I'm going to test it this afternoon. Hope that was the problem :)

MrSliff, 2016-12-22:
Worked, thank you!

closed 2016-12-22 by MrSliff