Traumflug/Teacup_Firmware #259: Z-Homing issue

MrSliff, 2016-12-22:
Hello,

I have some trouble getting the Z-Axis and Z_Min homing configured correctly in the Firmware.

X and Y Homing works well, but when sending a G161 Z or G28 Z0 the Z-Axis just moves one or some more steps down. after that i cant send commands to the printer anymore.

Tried several combinations of Z_MAX/MIN and travveling limits. I had no luck until now.

The endstops are shown correctly when i send a M119. 

Moving the Z-Axis up and down works before i send a z-homing command.

Wurstnase, 2016-12-22:
32.000.000 steps/m? or 3.200.000 steps/m?

MrSliff, 2016-12-22:
32.000.000 steps/m, I have a M5 Z-Axis with 1/128 stepping

Wurstnase, 2016-12-23:
Can you please change this to 3.200.000? I know this would be later wrong but I want to be sure that this is maybe a known bug.

Wurstnase, 2016-12-23:
Change Z-steps but nothing else! So please keep speed settings.

MrSliff, 2016-12-23:
Yeah that works! 

So, the definition of STEPS_PER_M_Z is a 32bit long uint. 32.000.000 should be in range.

Wurstnase, 2016-12-23:
Maybe you need more than 65.535 steps for acceleration. Are you on experimental branch?

MrSliff, 2016-12-23:
No, i just downloadad the main branch as a zip-file.

Wurstnase, 2016-12-23:
Very easy. There should be an icon for the git shell on your desktop. Open it and go in the folder where your current code is. 

As first command `git status`.

MrSliff, 2016-12-23:
fatal: Not a git repository (or any of the parent directories): .git

MrSliff, 2016-12-23:
Im in the parent Teacup folder where .gitattributes is found

Wurstnase, 2016-12-23:
No problem. 
`git init`
`git remote add origin teacup-URL`
`git fetch`
`git status`

MrSliff, 2016-12-23:
so teacup-URL is the URL to the repo?

Like:

https://github.com/Traumflug/Teacup_Firmware.git

MrSliff, 2016-12-23:
git fetch
fatal: 'teacup-URL' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

MrSliff, 2016-12-23:
Are you in Germany atm? I have to go to work, and i'm back at approx 15:30 german time. So, I have time again then. Sorry.

Wurstnase, 2016-12-23:
Ok, here we go. First save your current work.
1. we create a new local branch called "mrsliff"
2. we add all files to it
3. we commit with "my first commit"
```
git checkout -b mrsliff
git add .
git commit -m"my first commit"
```
Now we are in your personal branch.
The remote address is wrong. So let's change it. And add the correct one.
And lets fetch all branches now.
```
git remote remove origin
git remote add origin https://github.com/Traumflug/Teacup_Firmware.git
git fetch
```
Now we can rebase your branch with the master.
You will see some conflicts. Don't worry. 
```
git rebase origin/master
```
Looks like you made changes in pinio.c and greycode.c.
When you need this changes
```
git checkout --theirs pinio.c greycode.c
```
if not
```
git checkout --ours pinio.c greycode.c
```
after that, add your changes and continue with rebase
```
git add .
git rebase --continue
```
Now you are in your privat branch of "mrsliff" with the complete history of teacup and we can make later changes or rebase it to experimental. So when you finished this, please let me know. Then we can do some other git-stuff.

MrSliff, 2016-12-23:
ok i think im ready

Wurstnase, 2016-12-23:
Ok. First we create a new branch. Name it maybe testing_branch
```
git branch testing_branch
```
To change from your mrsliff-branch to testing_branch check it out.
```
git chechout testing_branch
```
Next I guess this issue could be solved by an already posted commit. We will cherry-pick this to your new created branch.
```
git cherry-pick f81ac3d852ab8c
```
https://github.com/Traumflug/Teacup_Firmware/commit/f81ac3d852ab8c3bd4a5e9 is a SHA-ID. Every commit has one. You can see them in gitk.

MrSliff, 2016-12-23:
How do i find the SHA-ID? There is just an input for SHA-1

MrSliff, 2016-12-23:
But i can find dda.c changes

Wurstnase, 2016-12-24:
You don't need to find the SHA. Take the number I give you.

MrSliff, 2016-12-24:
ok i think i've done that, i can see this commit with a yellow circle in gitk

Wurstnase, 2016-12-24:
And does homing working now with normal stepping rates on z?

MrSliff, 2016-12-24:
wait a moment, i did something wrong. i'm in mrsliff branch and have to git stash.

MrSliff, 2016-12-24:
how does that work in detail?

i switch to testing_branch and cherry pick. all changes are done immediately in the firmware?

Wurstnase, 2016-12-24:
yes. maybe you have some changes before. then you need to save them via stash or with a new commit of your changes.

In gitk you can see on top of the commits, the last commit and a green or red "commit". Green is added with `git add`, red are changed files.

Before changing branches with checkout you should check, if something changed. If it's clean `nothing to commit, working tree clean` you can change your branch or cherry-pick stuff. If not, 
```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
```
you can revert the change with `git checkout -- <file>`, add changes for a new commit or save your changes on a stash with `git stash`. You can get your stashed files back with `git stash pop`.

MrSliff, 2016-12-24:
i have errors now when i want to cimpile this. Although the .o files and other stuff are now generated in the main teacup directory not in build directory

do you understand german?
Here are the errors:

`rootbox@ubuntu:~/Dropbox/Teacup$ make
arm-none-eabi-gcc --specs=nano.specs --specs=nosys.specs -T cmsis-lpc1114.ld -Wl,--as-needed -Wl,--gc-sections  dda.o Makefile Makefile-common   -o dda
Makefile: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
make: *** Keine Regel vorhanden, um das Ziel „dda“, 
  benötigt von „build/teacup.elf“, zu erstellen.  Schluss.`

MrSliff, 2016-12-24:
you should be able to see the recent project files with the link posted in initial post

Wurstnase, 2016-12-24:
Just start with the download.

MrSliff, 2016-12-24:
ah got it, there was a copy of dda, which was named 'dda - kopie.c'

i think that was the problem. deleted and compiled. going to try flash and try now.

MrSliff, 2016-12-24:
It works!

Thank you very much for this!

Wurstnase, 2016-12-24:
Excellent!

We had a similar issue in the past. 

https://github.com/Traumflug/Teacup_Firmware/issues/189#issuecomment-265855591
@Traumflug could add it (maybe after X-mas) to experimental. https://github.com/Traumflug/Teacup_Firmware/commit/f81ac3d852ab8c3b

MrSliff, 2016-12-24:
Okay, happy christmas to you and thank you very much for your help!

Do you maybe know a good general calibration guide, especially for the z-axis in a p3steel printer

MrSliff, 2016-12-24:
o0h damn, i have not seen, that #define steps_per_m_z was at 3.200.000 :(

Wurstnase, 2016-12-24:
Slow down `#define SEARCH_FEEDRATE_Z        500`.

The Gen7Arm can handle up to approx. 115.000 steps/s.
At 32.000.000 steps/m this is ~ 3,5m/s or max 215 mm/min.

So changing `#define SEARCH_FEEDRATE_Z       150`.

Also `#define MAXIMUM_FEEDRATE_E       150`. 1000 is way too much.

MrSliff, 2016-12-24:
Hmm, okay.

I did some testing, because lowering the search feedrate didnt work (for now):

I turned the Driver resolution down to 1/16 and 1/8 and 1/4. Changed the steps/m accordingly.

Now the motors start to make strange noises. Measured the Step frequency with the Oscilloscope and have found out, that the Step frequency when searching the endstop is approx the double of normal moving feedrate.

Maybe this is the problem, that the z-axis feedrate then gets too high. I changed the search feedrate to 1  with 1/128 too, but that didnt help. Same behaviour.

Wurstnase, 2016-12-24:
You've change max feedrate of E. Not from Z!

Also you should could change `ENDSTOP_CLEARANCE_Z ` to 3. But when you set max feedrate of Z correct, it should work.

MrSliff, 2016-12-24:
OK now it works.

`#define STEPS_PER_M_X            640000
#define STEPS_PER_M_Y            640000
#define STEPS_PER_M_Z            32000000
#define STEPS_PER_M_E            1164105

/** \def MAXIMUM_FEEDRATE_X MAXIMUM_FEEDRATE_Y MAXIMUM_FEEDRATE_Z MAXIMUM_FEEDRATE_E
  Used for G0 rapid moves and as a cap for all other feedrates.
*/
#define MAXIMUM_FEEDRATE_X       6000
#define MAXIMUM_FEEDRATE_Y       6000
#define MAXIMUM_FEEDRATE_Z       100
#define MAXIMUM_FEEDRATE_E       150

/** \def SEARCH_FEEDRATE_X SEARCH_FEEDRATE_Y SEARCH_FEEDRATE_Z
  Used when doing precision endstop search and as default feedrate. No
  SEARCH_FEEDRATE_E, as E can't be searched.
*/
#define SEARCH_FEEDRATE_X        2000
#define SEARCH_FEEDRATE_Y        2000
#define SEARCH_FEEDRATE_Z        150`

closed 2016-12-24 by MrSliff

Wurstnase, 2016-12-24:
You should increase max feedrate for e. This can be up to 5900mm/min. Else your retract could be too slow.

MrSliff, 2016-12-24:
ok, i had it on 1000, followed your post and set it to 150.