Traumflug/Teacup_Firmware #263: simulator's --report-temptable option is broken

phord, 2017-02-15:
It currently shows zeros for all the calculated values. The breakage bisects to this commit:

    commit 678184b037476149fc9520ee22a7c6a477c45bea (HEAD, refs/bisect/bad)
    Author: Robert Konklewski <nythil@gmail.com>
    Date:   Thu Apr 21 11:55:27 2016

Probably the simulator is too dependent on some temperature function, but I haven't looked closely.

phord, 2017-02-15:
Good:
    ./sim --report-temptable|head
    ; Temperature sensor test -1
    ; Sensor 0
    0 631.25
    1 625.25
    2 619.25
    3 613.25
    4 607.25
    5 601.25
    6 595.25
    7 589.00

Bad:
    ./sim --report-temptable|head
    ; Temperature sensor test -1
    ; Sensor 0
    0 0.00
    1 0.00
    2 0.00
    3 0.00
    4 0.00
    5 0.00
    6 0.00
    7 0.00

Wurstnase, 2017-02-15:
https://github.com/Traumflug/Teacup_Firmware/commit/9948f5100ef7a666d3a5e9893e22070b7e21eb07
https://github.com/Traumflug/Teacup_Firmware/tree/sim_temp

I will rebase it to current experimental.

Wurstnase, 2017-02-15:
Maybe you have a better idea for a fix?