Traumflug/Teacup_Firmware #268: dda.c: update current position for corexy

Wurstnase, 2017-04-13:
`update_current_position()` is probably wrong currently for coreXY.
We need something like `corexy_to_cartesian()`.

Wurstnase, 2017-04-17:
I had some ideas to fix that. The most efficient (space) way I found is https://github.com/Traumflug/Teacup_Firmware/commit/bf41cc2d680afd.

This will cost some bytes for cartesian printers.
config|Program|Data
---|---|---
config.regtest-gen7-avr.h|32|0
config.regtest-ramps.h|32|0
config.regtest-nanoheart.h|80|0
config.regtest-teensy2.h|46|0
config.regtest-gen3.h|32|0
config.h.Profiling|46|0
config.regtest-display.h|80|0
config.regtest-acceleration-reprap.h|26|0
config.regtest-acceleration-temporal.h|26|0
config.regtest-gen7-arm.h|32|0
config.regtest-no-endstops.h|34|0
config.regtest-no-lookahead.h|26|0
config.regtest-cnc-shield-stm32.h|0|0

M114 on CoreXY prepatch:
[image]
M114 on CoreXY postpatch:
[image]

Wurstnase, 2017-04-18:
Simply recalculate only dedicated axes.
With `-flto` the space becomes a bit unpredictable.

config|Program|Data
---|---|---
config.regtest-gen7-avr.h|6|0
config.regtest-ramps.h|6|0
config.regtest-nanoheart.h|24|0
config.regtest-teensy2.h|-20|0
config.regtest-gen3.h|6|0
config.h.Profiling|-20|0
config.regtest-display.h|24|0
config.regtest-acceleration-reprap.h|8|0
config.regtest-acceleration-temporal.h|8|0
config.regtest-gen7-arm.h|4|4
config.regtest-no-endstops.h|6|0
config.regtest-no-lookahead.h|8|0

phord, 2017-04-19:
Do other firmwares update M114 coordinates during a move?  I guess with LCD support it can be very interesting. But I wonder if any external software uses this feature.

The change seems correct to me.

Wurstnase, 2017-04-19:
Two years ago Marlin hasn't this feature. It will only show the position where it should be when the planner will be empty. 

I like this M114 in Teacup. Also for debugging without hardware.

phord, 2017-04-19:
The simulator also calculates the position and will report it to the console, but it is only in steps and not in um.  So technically it is a step-wise position on corexy, though not the real axis steps.

750acb41bc3312d31140349e4b494394a133379c

Wurstnase, 2017-04-20:
Thanks for pointing.

closed 2017-04-20 by Wurstnase