Traumflug/Teacup_Firmware #269: Teacup Delta in UNO Board (atmega 328p - cnc shield v3) cant loging in Pronterface

caslor, 2017-05-08:
Hi 
i would like to make a small delta printer so i thought to use the Teacup firmware
i downloaded the Delta Branch firmware and have manage to upload it to my Arduino Uno
I am using a CNC Shield V3  with UNO board

after i have succesfull instalation of the firmware in my board when i am trying to connect to one of the programs  like pronterface , repetier   nothing happen..
in Pronterface i am getting the status that is connecting but stack there....  nothing going on..

i try with the  Teacup Master  firmware  and the  board connected fine..

what should i check in the Delta Firmware in order solve this bug ?

Thanks in Advance

ryannining, 2017-05-30:
i am working on delta teacup too (and cnc). i was merging manually the delta branch and the master branch. work in progress, movement is work, but havent test printing, still focus on CNC.

I make my own PCB board.

Traumflug, 2017-05-30:
Please don't merge. Merging makes branches un-handleable.

Better approach see informational issue #81.

ryannining, 2017-05-30:
yea, only on my pc, i wont upload it, no worry.

Traumflug, 2017-05-30:
> i wont upload it

Exactly this is my worry. If everybody had this attitude, Teacup wouldn't exist.

ryannining, 2017-05-30:
because i still not familiar with the git, i'm afraid i make mistake when upload.. so i should learn the git later, but meanwhile i will make sure all feature i want to add work. 

sorry....