Traumflug/Teacup_Firmware #271: git-step-rebase improvements

Traumflug, 2017-06-04:
While working on a much larger Git repo I could apply step-rebasing quite successfully. A branch some 1200 commits long (filtered down from 5200 commits) was rebased by 800 commits within a few days. People on the Git mailing list recommended to buy a book and applying the Mikado-method instead. A method which sounds like taking months :-)

Anyways, this work also resulted in a couple of improvements to *git-step-rebase*, which I've put into branch *git-step-rebase*. If there are no objections I'll forward this to *experimental* next week.

Wurstnase, 2017-06-04:
Looks great!

I'm looking forward to test this when I'm back at home again. My normal computer is pretty fast. But at work the old script becomes very slow.

Traumflug, 2017-10-17:
This stuff was moved to branch *experimental* in July already. Works indeed fine. Using it a log outside of Teacup, too.

closed 2017-10-17 by Traumflug

phord, 2017-10-17:
Nice tool.  I haven't looked at it before now.  I might play with it at work.

It reminds me of Michael Haggerty's [git-imerge](https://github.com/mhagger/git-imerge).  You might find that work interesting; I do, but I find it to be quite slow when I have sufficiently complicated changes to warrant trying it again.

Traumflug, 2017-10-17:
Yupp. This *git-imerge* thing looks much like a totally overengineered version of *git-step-rebase*.