Traumflug/Teacup_Firmware #273: Print result not smooth (delta printer)

ryannining, 2017-06-09:
[image]

[image]

After succesfully print using DELTA printer on Nano 328P, there still some problem. Looks like the motion is not smooth betwen gcode line segment. I dont know if this because gcode parsing not fast enough or, each line segment startup calculation is not fast enough.

But judging from other firmware such repetier can do this smoothly, there must be something we can do.

Tomorow will test on cartesian printer, check if this is just DELTA problem or not.

ryannining, 2017-06-09:
After check, looks not gcode parser problem, and queue is full.

Then after change acceleration from 300 to 8000, its become lot smoother.

Does this mean the acceleration lookahead doesnot  remove ramps up/down on crossing ?

Now printing the dragon again using accel=8000 .

Traumflug, 2017-06-09:
> Does this mean the acceleration lookahead doesnot remove ramps up/down on crossing ?

Depends on your lookahead settings.

ryannining, 2017-06-09:
Look ahead is turned on, thats only the setting we have right ? oh the jerk, i use 20.

ryannining, 2017-06-10:
Result accel = 300 vs 8000

[image]
[image]

Its better....

Wurstnase, 2017-06-10:
The yerk-setting in Teacup is mm/min. 20 is very low. In the simulator you will see no big diffences with low yerk or lookahead off. 

I would test this with ~1000 acceleration and ~1200 yerk.

ryannining, 2017-06-10:
Thx, yes its work, should i close this issue. So next, I am working on compatibility with Repetier Host (including the EEPROM manager).

closed 2017-06-12 by ryannining