Traumflug/Teacup_Firmware #275: void it away

Wurstnase, 2017-08-08:
Hi,

I'm currently reading a lot of stuff from ChibiOS and try to understand the code.
Often they have some unused variables for functions. So they use something like:
```
    void func(arg) {
      (void)arg;
    }
```
to suppress compiler warnings.

In that case I remember one #pragma in the [serial-avr.c](https://github.com/Traumflug/Teacup_Firmware/blob/master/serial-avr.c#L96-L125)

So simply ignore such warning with replacing
```
    //#pragma GCC diagnostic ignored "-Wunused-but-set-variable"
    uint8_t trash;
    //#pragma GCC diagnostic pop

    trash = UDR0;
```

with 

```
    (void)URD0;
```

Traumflug, 2017-08-08:
As long as there's still a read on URD0 happening: no problem. Without a read (or turning off the interrupt), the read interrupt happens over an over again.

Wurstnase, 2017-08-08:
Sure. I've read the disassembly. With -O0 it becomes shorter code. With -Os it will be the same code.

Traumflug, 2017-08-08:
You're aware that bobc did a ChibiOS port a few years back?

https://github.com/bobc/Teacup_Firmware

After looking at how step pins are operated I wasn't too enthusiastic, but maybe there are goodies I missed. Also had no performance measuring equipment back then, so can't say how much faster or slower it actually was.

Wurstnase, 2017-08-09:
> You're aware that bobc did a ChibiOS port a few years back?

No, thanks for pointing!

There are a lot of nice features in ChibiOS. E.g. virtual timers, so you can easily add timeouts for functions. Also a lot of nice interrupt features, preventing priority inversion. And so on.

While using an OS can decrease the performance, it shouldn't be so huge. Complete code is written for maximum speed, then size, then features etc...

My first goal isn't to rewrite Teacup. Maybe later. 😄