Traumflug/Teacup_Firmware #276: Configtool.py wont launch

SunixDev, 2017-08-10:
Hello, i'm getting this error witn python 2.7.13 
and no way to lauch teacup 
i'm on window 10 

>>> 
========= RESTART: C:\RepRap\firmware\Teacup_Firmware\configtool.py =========

Traceback (most recent call last):
  File "C:\RepRap\firmware\Teacup_Firmware\configtool.py", line 185, in <module>
    StartGui(getSettings())
  File "C:\RepRap\firmware\Teacup_Firmware\configtool\gui.py", line 615, in StartGui
    frame = ConfigFrame(settings)
  File "C:\RepRap\firmware\Teacup_Firmware\configtool\gui.py", line 72, in __init__
    self.pgPrinter = PrinterPanel(self, self.nb, self.settings)
  File "C:\RepRap\firmware\Teacup_Firmware\configtool\printerpanel.py", line 47, in __init__
    self.pgMech = self.registerPage(MechanicalPage, "Mechanical")
  File "C:\RepRap\firmware\Teacup_Firmware\configtool\printerpanel.py", line 59, in registerPage
    font = self.settings.font, **kwargs)
  File "C:\RepRap\firmware\Teacup_Firmware\configtool\mechanicalpage.py", line 50, in __init__
    sz.AddSpacer((10, 10), pos = (0, 0))
TypeError: Sizer.AddSpacer(): argument 1 has unexpected type 'tuple'
>>>

SunixDev, 2017-08-10:
I hope you can help me 
Thanks 
Regards

jbernardis, 2017-08-11:
This makes no sense to me.  Argument 1 to AddSpacer is and should be a tuple describing the size of the spacer.  This looks to me to be environmental.

Wurstnase, 2017-08-11:
Just checked master/experimental on Windows 10.

```
Python 2.7.12
wx 3.0.2.0
```

No issues.

SunixDev, 2017-08-11:
Thank you finnally worked with Python 2.7.12
wx 3.0.2.0 , i was using wxPython from pip i guess it was the problem

closed 2017-08-11 by SunixDev

Wurstnase, 2017-08-11:
I think, pip on wx don't work with Windows.

Just for other users who could have this problem. Here is a link how to install wx on other systems:
https://wiki.wxpython.org/How%20to%20install%20wxPython

mattgilbertnet, 2018-06-30:
Wurstnase: Those instructions for Mac OS lead to [this page](https://www.wxpython.org/pages/downloads/), which says to use pip. What's the alternative method for installing wx on Mac?

Wurstnase, 2018-06-30:
Probably you will find a working version here https://sourceforge.net/p/wxpython/activity/?page=0&limit=100#5957130734309d3bfa02f565

mattgilbertnet, 2018-06-30:
It looks like the best way to do this on Mac is with Homebrew, like:
`brew install wxpython`

If you've tried pip already, you may need to uninstall those libraries first.

PeregrineHomes, 2019-07-08:
I realize that this is an old issue but I'm having trouble with the solution. The link above takes you to a site that just says to install wxpython with pip. How do you install it without pip (on windows) if that's the problem?

Wurstnase, 2019-07-10:
On windows you can use pip to install the latest wxPython 4 version. The experimental branch should work with wxPython 4.

Vitaliiij, 2020-04-27:
Good afternoon, dear professionals. But doing everything according to the instructions, not what does not work out. Help me please.
I have installed window 10 x64, I installed it from the official site according to the instructions, python 2.7.18, python 2.7.12, python 2.7.11, python 2.7.0, everything is installed normally. I installed wxPython 3.0.2.0 and 2.8.12.1 for python 2.7 on them, not through pip, but with the wxPython3.0-win64-3.0.2.0-py27n.exe installer.
And I'm trying to open, by double-clicking the left mouse button, the configtool.py file in the C: \ Teacup_Firmware-master folder, but nothing happens. Please help me very much. I have assembled a 3D printer on Arduino UNO, but I can’t start it because I can’t upload the Teacup firmware. Thank you in advance for your reply.