Traumflug/Teacup_Firmware #277: Cannot launch Configtool.py

hauskagoagent, 2017-09-09:
Hey guys, I tried to launch the configtool with python (i definitely did installed python, wxPython, and Arduino), but I got some error messages as shown below:

> $ python configtool.py
> Traceback (most recent call last):
>   File "configtool.py", line 27, in <module>
>     from configtool.settings import Settings
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\settings.py", line 96
>     print "Unknown %s option: %s - ignoring." % (self.section, opt)
>                                             ^
> SyntaxError: invalid syntax

what should I do with this? btw, my environment is Win 10.

williamchenghu, 2017-09-10:
---PROBLEM UPDATE---

same error message shows up under Mac environment as well.
Besides that, I tried different version of Python (2.7.13 and 3.6.2), there's a warning message along with Python version detect under configtool.py. However the message is trying to notify people there's not a compatible version of wxPython for Python3, which is not true. wxPython 4.0.0b1 has dofferent versions for python 2.7, 3.4, 3.5, 3.6 (ref. https://pypi.python.org/pypi/wxPython). So i hacked the version detection to be >=4. That's where those error messages i pasted came in, however they seems having problem with default settings.

Wurstnase, 2017-09-10:
Configtool itself is for Python 2.7.
Python3 has in a lot of cases a different syntax.

E.g.
Python 2.7
print „something“
Python 3
print(„something“)

hauskagoagent, 2017-09-10:
hey @Wurstnase , thanks for your reply. Now I cleaned up 3 and re-installed 2.7, however new error messages show up:

> C:\Users\hauska\Documents\git\Dwarf_Firmware>python configtool.py
> Traceback (most recent call last):
>   File "configtool.py", line 185, in <module>
>     StartGui(getSettings())
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\gui.py", line 615, in StartGui
>     frame = ConfigFrame(settings)
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\gui.py", line 72, in __init__
>     self.pgPrinter = PrinterPanel(self, self.nb, self.settings)
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\printerpanel.py", line 47, in __init__
>     self.pgMech = self.registerPage(MechanicalPage, "Mechanical")
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\printerpanel.py", line 59, in registerPage
>     font = self.settings.font, **kwargs)
>   File "C:\Users\hauska\Documents\git\Dwarf_Firmware\configtool\mechanicalpage.py", line 50, in __init__
>     sz.AddSpacer((10, 10), pos = (0, 0))
> TypeError: Sizer.AddSpacer(): argument 1 has unexpected type 'tuple'

have any ideas?

Wurstnase, 2017-09-10:
Don’t install wx with pip.

https://github.com/Traumflug/Teacup_Firmware/issues/276

hauskagoagent, 2017-09-10:
@Wurstnase thanks a lot, as soon as i installed pre-compiled package, it works. (i did install x64 pack but throws error, x32 wxPython pack works as a charm.)

closed 2017-09-10 by hauskagoagent

Vitaliiij, 2020-04-27:
Good afternoon, dear professionals. But doing everything according to the instructions, not what does not work out. Help me please.
I have installed window 10 x64, I installed it from the official site according to the instructions, python 2.7.18, python 2.7.12, python 2.7.11, python 2.7.0, everything is installed normally. I installed wxPython 3.0.2.0 and 2.8.12.1 for python 2.7 on them, not through pip, but with the wxPython3.0-win64-3.0.2.0-py27n.exe installer.
And I'm trying to open, by double-clicking the left mouse button, the configtool.py file in the C: \ Teacup_Firmware-master folder, but nothing happens. Please help me very much. I have assembled a 3D printer on Arduino UNO, but I can’t start it because I can’t upload the Teacup firmware. Thank you in advance for your reply.