Traumflug/Teacup_Firmware #278: REQUEST : Using one analog input for multiple endstops

trustfm, 2017-09-18:
It would be nice to add this feature 
The implementation can be found here : 
http://tronixstuff.com/2011/01/11/tutorial-using-analog-input-for-multiple-buttons/
The same can be done for an keyboard (cross type) implementation.

I think the use of 3 ports for the endstops is an overkill for smaller boards like ardiono uno. 
What do you think ?
I have implemented a lot of projects with this tecnique and works really well.

Traumflug, 2017-09-18:
> It would be nice to add this feature

Certainly something you could implement and push it to the Teacup repo. Do you want write access?

> I think the use of 3 ports for the endstops is an overkill for smaller boards

For Teacup Firmware it's perfectly fine to use one pin for all endstops. Just define all three to the same pin.

trustfm, 2017-09-18:
> For Teacup Firmware it's perfectly fine to use one pin for all endstops. Just define all three to the same pin.
Yes but does the firmware distinguishes which endstop is pressed ? How would be able to do homing with just one common pin ?

Traumflug, 2017-09-18:
> does the firmware distinguishes which endstop is pressed ?

If you move the X axis and the common pin gets triggered, it was obviously the X endstop which caused the trigger. And yes, Teacup supports this.

trustfm, 2017-09-18:
I see ... by homing one by one the axis you can still use just one pin. 
Would be nice tho using this feature for keypads...
The i2c 16x2 lcd implementation gives me serial errors (just hangs) when i am enabling it on an arduino uno. I was planning to do this trick for adding a simple 5 button keypad...
Thank you for your time