Traumflug/Teacup_Firmware #279: Endstops broken

Wurstnase, 2017-10-26:
Hi,

I'm currently working on the endstops.
They become broken with https://github.com/Traumflug/Teacup_Firmware/commit/35926224145bd

Wurstnase, 2017-10-27:
Solved!

And should be now a little bit easier to understand what happens there :)
https://github.com/Traumflug/Teacup_Firmware/commit/301c5ed149d

Also saves ~12bytes.

Wurstnase, 2017-11-01:
While working on this I found another issue.

With https://github.com/Traumflug/Teacup_Firmware/commit/8a4cd52a68ad9 we force c_min to be set after rampup_steps. This is the right way, but in the homeing process it looks like this breaks something.

But it's not the issue. It's a symptom!

In the home.c we travel to -1000000 and then to +1000000. This is a total delta of 2,000,000.
Here we calculate dda->c_min (https://github.com/Traumflug/Teacup_Firmware/blob/master/dda.c#L332-L334)
Unfortunately 2,000,000 * 2400UL will overflow any time. So we get unpredictable c_min's.
Reducing 1,000,000 um to 894,784 um will solve this.

Patch will follow.

Wurstnase, 2017-11-01:
Please check https://github.com/Traumflug/Teacup_Firmware/tree/home_overflow

Wurstnase, 2017-11-01:
Maybe it's a better description that the last two moves shouldn't be greater than 32bit/2400.

Traumflug, 2017-11-01:
Why only `MAX_DELTA_UM / 2` ? I see no reason to not use `MAX_DELTA_UM`. In the extreme case, an endstop movement just as long as an unsigned integer allows, the endstop would trigger in the deceleration phase.

Wurstnase, 2017-11-01:
Because the total length becomes:
`(MAX_DELTA_UM / 2) - (-(MAX_DELTA_UM / 2))`

Check M111 S32 for this.

Wurstnase, 2017-11-01:
Oh dear...

Thx for pointing.
We don't need to travel from -MAX to +MAX.
When we have a min-endstop simply travel to -MAX and after that slowly back to 0.
When we have a max-endstop simply travel to +MAX and after that slowly back to 0.

Traumflug, 2017-11-01:
Now I understand your intention (M111 is 'set debug mode').

Yes, it's fair to assume that endstop search movements start at zero. To make sure this assumption holds, it can be explicitely to zero.

Wurstnase, 2017-11-01:
Can easily tested with e.g. z_min endstop:
```
G92 Z1000
G28 Z0
```

Wurstnase, 2017-11-04:
Refactoring home.c:

config|Program|Data
---|---|---
config.regtest-gen7-avr.h|-216|0
config.regtest-ramps.h|-176|0
config.regtest-nanoheart.h|-164|0
config.regtest-teensy2.h|-164|0
config.regtest-gen3.h|-216|0
config.h.Profiling|-164|0
config.regtest-display.h|-216|0
config.regtest-acceleration-reprap.h|-176|0
config.regtest-acceleration-temporal.h|-176|0
config.regtest-gen7-arm.h|-132|0
config.regtest-no-endstops.h|0|0
config.regtest-no-lookahead.h|-216|0
config.regtest-cnc-shield-stm32.h|-20|0

oli-obk, 2017-12-06:
Would it be possible to merge the experimental branch? It took me quite a while to figure out that endstops didn't work during homing.

Traumflug, 2017-12-06:
Teacup never merges, but I just forwarded *master* to a point past the endstop code changes. Thanks for the hint.

oli-obk, 2017-12-06:
Super! Thanks

Wurstnase, 2017-12-07:
@oli-obk please give some feedback, if everything is working properly again.

oli-obk, 2017-12-07:
@Wurstnase I'm still on experimental (which works fine). I'll try out master on the weekend.

Wurstnase, 2017-12-07:
Great. When experimental works, the current master should also work. 

Only different is, the e-axis could show wrong values while moving and using M114, on the master.

oli-obk, 2017-12-07:
I haven't even gotten to the e-axis yet xD I'm just happy that a print dry run without filament worked. But yea, it seemed like the extruder didn't do anything at all

Traumflug, 2017-12-07:
@Wurstnase, please don't hesitate to forward *master* on your own. You know much better what's currently going on.

The distinction between *master* and *experimental* was put into place to give a safe fallback (*master*) for new users, while allowing to test new features for users knowing about them. Branch *experimental* as kind of a final version of feature branches. Quite a number of times I committed pure bug fixes to *master* instead of *experimental*.

Maybe this strategy no longer applies. Maybe *experimental* is no longer needed. I don't know.

Wurstnase, 2017-12-07:
@Traumflug I will do it next time. I was on a travel while doing this things.

Experimental is still a good idea. I will use this branch in future more frequently. New/standard users are afraight of others branches. So _master_ and _experimental_ will always do this job perfectly for them.

Wurstnase, 2017-12-07:
> But yea, it seemed like the extruder didn't do anything at all

Oooo... Nothing? Ah wait. Yes, I fixed this also. I will push it.

Wurstnase, 2017-12-07:
Please test it. I pushed it also to master. My last testing was successfull.
Most issues were implemented some month ago. But you and only one other guy were falling over this.

oli-obk, 2017-12-10:
The extruder moves if explicitly commanded so by pronterface, but during printing, nothing happens.

I do get a few bad M codes, so it might be something in my tooling though:

```
E: Bad M-code 141
E: Bad M-code 142
E: Bad M-code 113
E: Bad M-code 108
```

Wurstnase, 2017-12-10:
You could upload your gcode with gist. so I could check your code. How do you create the gcode? In Slic3r e.g. you can explicitly teacup as gcode version.