Traumflug/Teacup_Firmware #280: Broken Serial Communication

YaseenTwati, 2017-10-31:
I Built my own custom Controller PCB around the **ATMEGA328P** running at 20Mhz
i can successfully build the firmware and upload it, but serial communication doesn't seem to work work
i basically receive garbage on the computer side ( same sort  of garbage you'd get from a wrong baud rate )
i tried multiple baud rates, no success, 
tried running my board on its internal 8Mhz, same problem
tried using a simple arduino sketch to print "Hi" just to make sure that serial works .. and it does
tried different adapters just in case ( serial to usb and the uart bluetooth module )
i couldn't get this to work
lastly but not least 
i tried  uploading the firmware to a bare arduino uno which isnt connected to anything just to check the serial response .. same exact problem.
am i missing something ?

YaseenTwati, 2017-10-31:
tried all lol
Arduino IDE
config tool
Makefile

Wurstnase, 2017-10-31:
Just doing some tests some days ago with the master and experimental on an atmega2560 at 16MHz. No issue at all. 

Someone on an old Gen7 1.5 with 20MHz had also no problems.

YaseenTwati, 2017-10-31:
@Wurstnase 
Weird, its probably something with my setup on the software/buid side since the hardware seems to be okay
just couldn't get what it is ...
if you have time, is there anything weird about my board config and Makefile ? 

[Makefile.txt](https://github.com/Traumflug/Teacup_Firmware/files/1431789/Makefile.txt)

[board.spiro.h.txt](https://github.com/Traumflug/Teacup_Firmware/files/1431792/board.spiro.h.txt)

YaseenTwati, 2017-10-31:
ill edit the code to generate a square wave signal and check it with my oscilloscope
to check if the code is running at the correct frequency ..

Wurstnase, 2017-10-31:
I can't find any special. 
Only your baudrate looks like pretty slow. But that shouldn't be the problem. 

How does your Arduino test sketch looks like?
How is the schematic of your board?

YaseenTwati, 2017-10-31:
After messing around and doing some Oscilloscope tests
the first test failed which means code was running at wrong frequency 
i went through changing the fuses again and it worked this time .. even though i changed the many times earlier ...

[image]

[image]
[image]

conclusion : problem with my fuses and not teacup
sorry for the trouble, 
Thanks for the help

closed 2017-10-31 by YaseenTwati