Traumflug/Teacup_Firmware #281: WX problem on new version python+wx

ryannining, 2017-11-02:
After install latest 2.7.14 python and using PIP to get WXpython. Need to change configtool code

AddSpacer have been replaced with just Add.
wx.OPEN change to wx.FD_OPEN
wx.CHANGE_DIR to wx.FD_CHANGE_DIR

windows 32bit

Wurstnase, 2017-11-02:
Can you please check the binary installer of wx for windows?

Wurstnase, 2017-11-02:
I see. There is a new version from Phoenix. Wx 4.0.0b2. 

I will check this.

ryannining, 2017-11-03:
I do pip uninstall 4.0 and install binary 3.0 its work again.

Wurstnase, 2017-11-03:
With wx 4.0 alot things changed. Some functions were renamed or starts to be deprecated.
I will take a look into it how easy it is to fix that for the future.

ryannining, 2017-11-07:
I have edit the python file to work on wx 4, and i uninstall the wx4 then install wx3 binary. I am suprised, that it still work.

Wurstnase, 2017-11-08:
Good news.
Maybe @traumflug can add you and you can upload a new branch with your
working version.

ryannining, 2017-11-08:
I have upload on https://github.com/ryannining/Teacup_Firmware , is it okay ?

Traumflug, 2017-11-08:
> Maybe @traumflug can add you

He is collaborator already :-)

> I have upload on https://github.com/ryannining/Teacup_Firmware

Dealing with foreign repos is usually hard. Ideally we get a branch on top of *experimental* with proper indentation (2 spaces), clutter removed and [atomic commits](https://seesparkbox.com/foundry/atomic_commits_with_git). `git rebase -i` is a great tool.

Wurstnase, 2018-12-21:
I'm working on this issue. Please check the wx4 branch.
https://github.com/Traumflug/Teacup_Firmware/tree/wx4

Wurstnase, 2018-12-21:
Btw. the configtool and other scripts should now work with Python 2.7 and Python 3.6+.