Traumflug/Teacup_Firmware #283: Compile error for board.gen7-arm.h with experimental branch

Capo01, 2017-11-04:
I see the following error when attempting to compile the experimental branch for gen7-arm board.  The error is the same when compiling on both windows and Ubuntu. 

> $ make
>   CC        build/analog.o
> In file included from config.h:3:0,
>                  from config_wrapper.h:37,
>                  from analog-stm32.c:56,
>                  from analog.c:41:
> config\board.gen7-arm.h: In function 'init_analog':
> config\board.gen7-arm.h:181:6: error: 'PIO1_1_PORT' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(extruder, TT_THERMISTOR, PIO1_1,THERMISTOR_EXTRUDER)
>       ^~~~~~~~~~~
> config\board.gen7-arm.h:181:6: note: each undeclared identifier is reported only once for each function it appears in
> config\board.gen7-arm.h:181:39: error: 'PIO1_1_PIN' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(extruder, TT_THERMISTOR, PIO1_1,THERMISTOR_EXTRUDER)
>                                        ^~~~~~~~~~
> config\board.gen7-arm.h:182:6: error: 'PIO1_0_PORT' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(bed,      TT_THERMISTOR, PIO1_0,THERMISTOR_BED)
>       ^~~~~~~~~~~
> config\board.gen7-arm.h:182:39: error: 'PIO1_0_PIN' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(bed,      TT_THERMISTOR, PIO1_0,THERMISTOR_BED)
>                                        ^~~~~~~~~~
> In file included from config.h:3:0,
>                  from config_wrapper.h:37,
>                  from analog-stm32.c:97,
>                  from analog.c:41:
> config\board.gen7-arm.h:181:42: error: 'PIO1_1_ADC' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(extruder, TT_THERMISTOR, PIO1_1,THERMISTOR_EXTRUDER)
>                                           ^~~~~~~~~~
> config\board.gen7-arm.h:182:42: error: 'PIO1_0_ADC' undeclared (first use in this function)
>  DEFINE_TEMP_SENSOR(bed,      TT_THERMISTOR, PIO1_0,THERMISTOR_BED)
>                                           ^~~~~~~~~~
> make: *** [build/analog.o] Error 1

Capo01, 2017-11-04:
I see the missing definitions in Teacup_Firmware/arduino_lpc1114.h. However, I am unsure as to how I should include this file and where.

Thanks

Wurstnase, 2017-11-04:
Please check the definition in the Makefile-ARM.

MCU ?= lpc1114

Wurstnase, 2017-11-04:
Please check also the latest experimental. I've changed the standard to lpc1114 in the Makefile-ARM. 

There are some homing issues in the master-branch which are should solved in the experimental-branch.

Capo01, 2017-11-04:
I cloned the latest experimental-branch and it now complies without error. Thanks for your help!

closed 2017-11-04 by Capo01