Traumflug/Teacup_Firmware #286: Stuttering movement on high polygon object

ryannining, 2017-11-22:
I was printing this test object  And the x y movement is very stuttering. So i investigate and found this piece of code fix that and make the high polygon smooth. But, i dont know exactly what happen, removing Z and E from calculating max_factor_speed does eliminate the problem, but it will make the z movement (change layer) is very sudden.

My step/mm X:112 Y:112 Z:400 E:420, cartesian printer.

[tube.zip](https://github.com/Traumflug/Teacup_Firmware/files/1496541/tube.zip)

dda_lookahead.c on func dda_find_crossing_speed

from
`  for (i = X; i < AXIS_COUNT; i++) {
`
to

`  for (i = X; i < E; i++) {
    if ((prev->delta[i]==0) && (current->delta[i]==0)) continue;  `
Thx

Wurstnase, 2017-11-30:
Here is a more generic way:

```
@@ -77,8 +77,14 @@ void dda_find_crossing_speed(DDA *prev, DDA *current) {
   //       Caveat: bail out condition above and some other non-continuous
   //               situations might need some extra code for handling.
   for (i = X; i < AXIS_COUNT; i++) {
-    prevF[i] = muldiv(prev->delta[i], F, prev->total_steps);
-    currF[i] = muldiv(current->delta[i], F, current->total_steps);
+    if (!prev->delta[i])
+      prevF[i] = muldiv(prev->delta[i], F, prev->total_steps);
+    else
+      prevF[i] = 0;
+    if (!current->delta[i])
+      currF[i] = muldiv(current->delta[i], F, current->total_steps);
+    else
+      currF[i] = 0;
   }
 
   if (DEBUG_DDA && (debug_flags & DEBUG_DDA))
```
While often the E-Axis is much slower it should also calculated. So when you print not a spiral vase, this part of code should become much faster.

ryannining, 2017-12-01:
if we change prevF[i] and currF[i] into just prevF and currF it should save same ram right ? merge 2 for into 1, and remove the

Traumflug, 2017-12-01:
Nope. This is a per-axis calculation.

Doing this calculation with the combined feedrate whouldn't take direction changes into account.