Traumflug/Teacup_Firmware #287: Fix extruder moving when start printing

ryannining, 2017-12-01:
Condition: If we play with extruder feed button in repetier and then start Print a file/gcode with some initial Gcode movement before G92 E0
Sample gcode:
```
;Generated with Cura_SteamEngine 15.01
; Default start code
G28 ; Home extruder
G1 Z15 F4000
M107 ; Turn off fan

G90 ; Absolute positioning
M82 ; Extruder in absolute mode
; Activate all used extruder
;M104 T0 S210
G92 E0 ; Reset extruder position
```

the `G1 Z15 F4000` will cause the extruder moving. To fix can be fix the initial Gcode, move the problem code after the G92 E0, or fix teacup code in gcode_process.c in G28 code, to run G92 E0

```
//G92 E0 after G28
startpoint.axis[E] = next_target.target.axis[E] = 0;
dda_new_startpoint();

```

after the `home();`

thats fix the problem in my case, .

Wurstnase, 2017-12-01:
Sorry, maybe a little bit to direct, but this is not a fix. This is a hack. 

The more interesting point: why it's moving when no command is sent for that axis?

ryannining, 2017-12-02:
i will check again

Traumflug, 2017-12-02:
Dunno how skilled you're with debugging. Usually it's a good idea to put a lot of `sersendf_P()` into the preparing code to check assumptions. There's quite some code in *dda.c* already, showing how this is done. Formatting options can be seen in *sendf.c*. Output goes to the serial line, so it's visible in Pronterface with *Debug G-code* enabled, see menu there.

With a single movement, or the first movement of a series, lookahead can't be guilty, because it's ignored. If the extruder moves unexpectedly, a likely reason is that the step counter isn't zero. Something one can check with such debugging code.

Wurstnase, 2017-12-02:
Currently I found some strange behavior when switching between G91 and G90 and moving the E-axis.
Maybe you can verify this?

Wurstnase, 2017-12-02:
The code above seems to work.
But not this code:
```
G91
G1 E10
G1 E10
G90
G1 E10
```

Wurstnase, 2017-12-02:
I guess my issue is different from yours.
I enabled E_ABSOLUTE and DEBUG:
```
M111 S32
G91
G1 E10 F500
Create: X 0.000  Y 0.000  Z 0.000  F 500
[0,0,0,10000] [ts:3200,ds:10000] }
Start: X 0.000  Y 0.000  Z 0.000  F 500
; here I wait until done
G1 E10 F500
Create: X 0.000  Y 0.000  Z 0.000  F 500
[0,0,0,0] [ts:3200,ds:10000] }
Start: X 0.000  Y 0.000  Z 0.000  F 500
; wait until done
G90
G1 E10 F500
Create: X 0.000  Y 0.000  Z 0.000  F 500
[0,0,0,0] [ts:3200,ds:0] }
Start: X 0.000  Y 0.000  Z 0.000  F 500
```
Last ds is zero, which is not correct.

Wurstnase, 2017-12-02:
Looks like this could fix it.
```
@@ -461,6 +463,8 @@ void dda_create(DDA *dda, const TARGET *target) {
 
     // next dda starts where we finish
     memcpy(&startpoint, &dda->endpoint, sizeof(TARGET));
+    if (startpoint.e_relative)
+      startpoint.axis[E] = 0;
     #ifdef LOOKAHEAD
       prev_dda = dda;
     #endif
```

Caveat: In the end of each move E is zero in M114.
I would split pure relative E-axis and all-relative.
So with M83 E-axis becomes relative with that behavior.
But with G91 (all relative) and M82 (e normally absolute), E will stay absolute. Only movement is now relative. So switching between will work.

E.g:
```
M83
G1 E10
M114 -> E0.000
M82
G91
G1 E10
M114 -> E10.000
G1 E10
M114 -> E20.000
G90
G1 E10 -> movement from E20 to E10
M114 -> E10.000
```

This would need a change in gcode-process.c also.

Wurstnase, 2017-12-02:
Please check: https://github.com/Traumflug/Teacup_Firmware/tree/issue_287

Wurstnase, 2017-12-07:
Should be fixed in master. Please reopen when you still have problems.

closed 2017-12-07 by Wurstnase