Traumflug/Teacup_Firmware #289: UNO Enable INPUT_PULLUP for TEMP_THERMISTOR

rascal71, 2018-01-11:
Hello 
I'm building a mini-printer with some 28byj-48 steppers with a vanilla arduino UNO board using the ULN2003 Darlington array IC driver boards. Takes four pins per stepper to drive them. I'm using all pins counting one each for extruder temperature and heat. (no heated bed) I have all the axis and extruder working so far.

I want to enable the INPUT_PULLUP resistor for reading the thermistor properly, otherwise it reads 0 or 1023. Only way I could get it reading correctly is by adding this here:

/**
Start a new ADC conversion.
*/
void start_adc() {
  PULLUP_ON(AIO4); // why here?
  ADCSRA |= MASK(ADSC);
}

This gets called often by the clock, it works but there must be a better way? I tried in pinio.c but that wasn't working, I can't seem to figure out where this gets set or unset I guess. Thanks!

Traumflug, 2018-01-11:
> I want to enable the INPUT_PULLUP resistor for reading the thermistor properly, otherwise it reads 0 or 1023.

Are you sure your circuitry is working properly? Pullup resistors can help with digital signals, but usually the last thing one wants is to have them on an analog input, messing the reading up. One should be able to measure this analog signal with a voltage meter directly on the chip pin.

If the circuitry is fine and it still looks like you need a pullup, something else is wrong. Wrong reference voltage choosen, for example.

triffid, 2018-01-11:
For the thermistor, you need an external 1k-4k7 pullup to AVCC. The
internal pullup is far too weak to give a usable analog range.

See
http://hydraraptor.blogspot.com/2007/10/measuring-temperature-easy-way.html
for a discussion that should give some hints as to why an internal pullup
of ~100k(ish) is not a good choice.

It also suggests that self-heating may be a significant issue with 5v,
consider using the 1.1v internal bandgap and an op-amp voltage follower
from AREF to the high side resistor - but this added complication may be
more trouble than it's worth if you're new to electronics

rascal71, 2018-01-21:
Doesn't using the internal pull-up essentially accomplish the same thing as an external resistor? The internal pull-up is ~20k when building an external circuit with two 10k resistors in series (I don't have any 20k) I get the same results as using the pull-up. I've regenerated the thermistor lookup table using the 20k value and r2 from the data sheet and everything appears to read and work correctly.

Traumflug, 2018-01-21:
> Doesn't using the internal pull-up essentially accomplish the same thing as an external resistor?

Yes. But it's an analog signal, so it's essential to have not any, but a well defined resistor in place. For best results, one should even use precision resistors, even if virtually no controller hardware has them.

Changing the lookup table to match the MCUs internal resistor is also a way to go, of course. If it works and gives reasonable measurements (good references: melting ice, room temperature, boiling water), everything is fine.