Traumflug/Teacup_Firmware #290: Arduino compiling error when enabling I2C HD44780 DISPLAY

anmthu008, 2018-01-21:
hey guys.. why do i get this error when compiling whenever I enable DISPLAY BUS I2C and HD44780/1602? Cuz i intend to use my 1602 LCD with i2c and connecting SDA/SCK to respective A4/A5 pins to the board.

```
In file included from C:\Users\AUNGNA~1\AppData\Local\Temp\arduino_build_258095\sketch\display.c:26:0:

C:\Users\AUNGNA~1\AppData\Local\Temp\arduino_build_258095\sketch\display_hd44780.c: In function 'display_init':

display_hd44780.c:30: error: 'parallel_4bit_instruction' undeclared (first use in this function)
   displaybus_write(0x06, parallel_4bit_instruction);
                          ^

C:\Users\AUNGNA~1\AppData\Local\Temp\arduino_build_258095\sketch\display_hd44780.c:30:26: note: each undeclared identifier is reported only once for each function it appears in

C:\Users\AUNGNA~1\AppData\Local\Temp\arduino_build_258095\sketch\display_hd44780.c: In function 'display_tick':

display_hd44780.c:95: error: 'parallel_4bit_instruction' undeclared (first use in this function)
         displaybus_write(0x01, parallel_4bit_instruction);
                                ^

display_hd44780.c:127: error: 'parallel_4bit_data' undeclared (first use in this function)
         displaybus_write(data, parallel_4bit_data);
                                ^
exit status 1
'parallel_4bit_instruction' undeclared (first use in this function)
```

FYI, i didnt modified the code in anyway, I just downloaded the firmware and try to compile for my UNO board with CNC SHIELD V3. all with default settings (except the printer motor settings and thermistor settings)

Traumflug, 2018-01-21:
You're probably the first to try the combination of I2C bus and HD44780 display. Code is designed to allow this combination, but so far nobody tried. For development we had a parallel wired HD44780 and a graphical I2C display with an entirely different protocol.

Error messages hint to some `#ifdef` missing or in excess. If you manage to fix this: cheers to you! And double-cheers for providing a patch!

anmthu008, 2018-01-22:
@Traumflug , may i know which i2c display you guys tested with?

Traumflug, 2018-01-22:
Sure enough, a parallel bus needs more pins. But that's what most people have in hardware and Teacup should work on any hardware. Accordingly the strategy choosen is to have modules for each, communications bus and the display its self. Similar to C++ classes.

I2C display development can be seen here: https://github.com/Traumflug/Teacup_Firmware/issues/190 . It's a SSD1306 type display. Some hints on development of the HD44780 display are here: https://github.com/Traumflug/Teacup_Firmware/issues/214

anmthu008, 2018-01-22:
alright.. thanks @Traumflug