Traumflug/Teacup_Firmware #297: Arduino UNO CNC Shield V3

anmthu008, 2018-05-10:
Hey guys, is why isnt the stepper motors holding torque as soon as it finishes the movement in Arduino UNO with CNC Shield V3 (A4988 drivers)... For e.g  if i print a box and when X is moving, y wont be moving (as if turned off, not holding).. This leads to quite alot of inaccuracies..  Is there any solution to this (like how we can enable and disable stepper motors like in Marlin).. btw, i've uploaded my config for reference..  thx guys

[board.cnc-shield-v3.txt](https://github.com/Traumflug/Teacup_Firmware/files/1993445/board.cnc-shield-v3.txt)
[printer.uno.txt](https://github.com/Traumflug/Teacup_Firmware/files/1993446/printer.uno.txt)

Wurstnase, 2018-05-11:
Hi,

First, you've enabled z auto disable.

Second, you setup all enable pins to the same output. While this is true in
the CNC shield, you add the z disable after each movement to all axis.

In your printer.xxx.h comment out the auto disable. And in the board.xxx.h
comment out all enable pins and delete the comment for stepper enable pin.

anmthu008, 2018-05-15:
Ahh i c... no wonder.. Thanks @Wurstnase .. just did it and it fixed everything.. cant believe i've wasted about a week on fixing this.. lol

closed 2018-05-15 by anmthu008