Traumflug/Teacup_Firmware #301: Tool pids affected by bed pids

phord, 2018-07-08:
I noticed if the bed heater is maintaining a setpoint then the tool heater drifts by +/- 5 degrees.  But if the bed heater is solid-on or solid-off, the tool temperature only drifts half a degree.  Why are the correlated?  Is it real or is it a read inaccuracy?

[image]

triffid, 2018-07-08:
power supply noise, or poor load regulation?

Wurstnase, 2018-07-08:
I had some years a similar issue. In my case was it a bad layout from the shield. A switching regulator was close to the ADC line.

phord, 2018-07-10:
Could be.  Cheap-o RAMPS 1.4.  I guess I know even better why we have TEMP_EWMA now.  :imp:

phord, 2018-07-10:
Hm.  Obviously I should move my Gen7 board over since I no longer use the printer it is attached to.  And because this RAMPS 1.4 [may be a fire hazard](https://reprap.org/wiki/RAMPS_1.4#WARNING_-_THERMAL-isolation-related_DESIGN_FLAW_IN_Power_handling_capacity_of_PRODUCTION_RAMPS_1.4_boards).

closed 2018-07-18 by phord